nodemcu/nodemcu-firmware #1780: tmr.delay example in documentation may be misleading

marcelstoer, 2017-02-05:
The example at http://nodemcu.readthedocs.io/en/latest/en/modules/tmr/#tmrdelay delays the execution for 100us. Yet, I believe that value shouldn't ever be beyond 50us, correct? Was it you @TerryE from whom I picked that up?

TerryE, 2017-02-06:
the delay function our uses  `os_delay_us` which just maps onto the ROM function `ets_delay_us`.  If you look at the ROM code this is simply a loop which polls the CPU xtal using the `rsr.ccount` instruction.  Most importantly, this loop does not mask interrupts so ISRs can still be delivered and hence the 50 uSec limit does not apply in this case.  However, there is still the wider guideline from the SDK that no user task should run for more than 15 mSec (otherwise system tasks relating to the network stack might start to fail).

marcelstoer, 2017-02-07:
> the 50 uSec limit does not apply in this case

Thanks Terry, in which case(s) _does_ it apply?

TerryE, 2017-02-07:
>  in which case(s) does it apply?

Essentially anywhere the code disables interrupts. Just do a `grep -Ri intr` to see what I mean: dht, key, gpio, i2c_master, onewire, pwm spi, uart, ws2801, ws2812 all dick around with the interrupts, but most stay within the 50 uS guidelines. 

 `ws2801.writergb()` is an example that doesn't.  This doesn't mean that writing out a long string will cause the ESP8266 to crash, but masking off interrupts for this length of time can cause data drop on the UART, and the network stack to fail, and this in turn could confuse the application or cause the CPU to panic.

closed 2017-02-07 by marcelstoer