nodemcu/nodemcu-firmware #1781: http.get() looses right callback reference

Modellfan, 2017-02-05:
I had so much hope for the new released feature for the http.get to chain requests, but as it seems it doesn't work

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
to get two different requests at the nearly the same time (order doesn't play a role)
### Actual behavior
the callbacks of the two different requests seem two mix up internally. Also tried with node.task.post, no chance to get over it. Seems as the queue looses order or content for some reason.
### Test code
```
function shortip()
http.get("http://httpbin.org/ip", nil, function(code, data)
  if (code < 0) then
    print("HTTP request failed")
  else
    print("Short IP", code, data)
  end
end)
end

function longip()
http.get("http://ip-api.com/json", nil, function(code, data)
  if (code < 0) then
    print("HTTP request failed")
  else
    print("Long IP", code, data)
  end
end)
end

tmr.alarm(1, 3000, tmr.ALARM_AUTO, function() longip() end)
tmr.alarm(2, 3000, tmr.ALARM_AUTO, function() shortip() end)
```

Output:
` Short IP	200	{
  "origin": "xx"
}

Short IP	200	{
  "origin": "xx"
}

Short IP	200	{"as":"xx BELGACOM-SKYNET-AS","city":"Brussels","country":"Belgium","countryCode":"BE","isp":"Proximus Skynet","lat":50.8476,"lon":4.3428,"org":"Proximus Skynet","query":"xx","region":"BRU","regionName":"Brussels Capital","status":"success","timezone":"Europe/Brussels","zip":"1012"}
Short IP	200	{"as":"x BELGxACOM-SKYNET-AS","city":"Brussels","country":"Belgium","countryCode":"BE","isp":"Proximus Skynet","lat":50.8476,"lon":4.3428,"org":"Proximus Skynet","query":"xx","region":"BRU","regionName":"Brussels Capital","status":"success","timezone":"Europe/Brussels","zip":"1012"}
Short IP	200	{"as":"xx BELGACOM-SKYNET-AS","city":"Brussels","country":"Belgium","countryCode":"BE","isp":"Proximus Skynet","lat":50.8476,"lon":4.3428,"org":"Proximus Skynet","query":"xx","region":"BRU","regionName":"Brussels Capital","status":"success","timezone":"Europe/Brussels","zip":"1012"}
`

The result with the city name should be written with "Long IP" in advance. (ip anonimized with xx)

### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 20a591f59aee69f447b85f7345dc59a62b68e690
	SSL: false
	modules: adc,cjson,dht,file,gpio,http,i2c,mdns,mqtt,net,node,ow,pwm,rtctime,sntp,spi,tmr,uart,wifi
 build 	built on: 2017-02-05 18:34
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

8<------------------------ END BUG REPORT -------------------------------------

Modellfan, 2017-02-05:
Modified with node.task.post, same result:

`
function shortip()
http.get("http://httpbin.org/ip", nil, function(code, data)
node.task.post(function()
  if (code < 0) then
    print("HTTP request failed")
  else
    print("Short IP", code, data)
  end
end)
end)
end

function longip()
http.get("http://ip-api.com/json", nil, function(code, data)
node.task.post(function()
  if (code < 0) then
    print("HTTP request failed")
  else
    print("Long IP", code, data)
  end
end)
end)
end

tmr.alarm(1, 3000, tmr.ALARM_AUTO, function() longip() end)
tmr.alarm(2, 3000, tmr.ALARM_AUTO, function() shortip() end)`

Result:
`> Short IP	200	{"as":"xx BELGACOM-SKYNET-AS","city":"Brussels","country":"Belgium","countryCode":"BE","isp":"Proximus Skynet","lat":50.8476,"lon":4.3428,"org":"Proximus Skynet","query":"xx","region":"BRU","regionName":"Brussels Capital","status":"success","timezone":"Europe/Brussels","zip":"1012"}
Short IP	200	{
  "origin": "xx"
}
`
When I comment out the shortip() timer everything is working as expexted and with the longip() timer vise versa as well. So it must be for some reason, i can't fully explain the coexistance of two http.get.

Thank the developers for their great work so far.

pjsg, 2017-02-06:
Ah yes -- this isn't the use case that was fixed in the latest code. The case that was fixed is that you can now do:
```
http.get("http://httpbin.org/ip", nil, function(code, data)
  if (code < 0) then
    print("HTTP request failed")
  else
    print("Short IP", code, data)
  end
  http.get("http://ip-api.com/json", nil, function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print("Long IP", code, data)
    end
  end
end
```

I.e. you can now call another get from the callback of the previous get. This makes certain things easier.

Yes, it would be great to be able to have multiple get's running at the same time. However, the httpclient implementation is somewhat obscure.

We would welcome a PR to this end.

marcelstoer, 2017-02-06:
This is not a bug. As per the [docs](http://nodemcu.readthedocs.io/en/latest/en/modules/http/):

> It is not possible to execute concurrent HTTP requests using this module.

closed 2017-02-06 by marcelstoer

karrots, 2017-02-08:
@Modellfan you need to wrap the whole `http.get` in a `node.task.post` not just the callback.

Modellfan, 2017-02-08:
```
http_blocked = false
http_blocked_time = tmr.now() --needed if http.get dies in blocked state
http_buff = nil

function http_get_save(str, retfunc)
  if (http_blocked_time+1000000*15) < tmr.now() then
    http_blocked = false
  end
  if not http_blocked then
    http_blocked = true
    http_blocked_time = tmr.now()
    http.get(str,nil, function(code,data)
      http_blocked = false
      retfunc(code,data)
      if http_buff then
        dprint("\n\tDoing something from Buffer")
        tempa = http_buff.retfunc
        tempb = http_buff.str
        http_buff=nil
        http_get_save(tempb,tempa)
      end
    end)
  else
    dprint("\n\tPut something in http_get Buffer")
    if http_buff then dprint("\tBuffer Overflow") end
    http_buff = {}
    http_buff.str = str
    http_buff.retfunc = retfunc
  end
end
```

Did the trick. Working since 48 hours very robust. I hope it helps.

devyte, 2017-02-08:
If this makes concurrent http.get()s work, I suggest adding it to the
examples, it would be most helpful to have it available.