nodemcu/nodemcu-firmware #1782: gpio.trig LOW triggers with HIGH level and travels in time

Uinden, 2017-02-08:
Hi.
I'm new to NodeMCU, ESP8266 and Lua so this report may be wrong. Tell me if it is, i will close it.
Learning triggers i got some strange results and after investigation i got little code to reproduce it.
## Hardware ##
**ESP-12-E**
**1kOhm** potentiometer
**10kOhm** resistor 
**VCC** is **3.41V** 
Connected like this (sorry for my mspaint skills):
[image]
## Software ##
I built mine **NodeMCU** online at https://nodemcu-build.com/ 
I used **integer**: http://nodemcu-build.com/builds/nodemcu-master-14-modules-2017-02-07-12-01-52-integer.bin
Load output is:
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
	SSL: false
	modules: adc,cjson,dht,file,gpio,i2c,net,node,ow,pwm,spi,tmr,uart,wifi
 build 	built on: 2017-02-07 12:00
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
```
## gpio.trig LOW triggers with HIGH level ##
Code:
```lua
pin = 1
last_triggered = 1
gpio.mode(pin,gpio.INT)
function pin1cb(level, time)
    print( level, time )
end
gpio.trig(pin, "low", pin1cb)
```
**GPIO5** shorted to **GND** produces:
```
0	53983615
0	53985692
0	53987772
0	53989884
0	53991961
0	53994041
0	53996173
0	53998250
```
It floods lots of triggers, but i assume it is okay. And level is **0**, and we wait for **LOW** so it is working.
Then i connect **GPIO5** like on picture above.
Slowly turning potentiometer until **GPIO5** have about **1.75V** starting to produce strange output.
```
0	322657341
0	322659424
0	322661508
0	322663592
0	322665675
0	322667759
0	322669846
0	322671971
0	322674055
0	322676138
0	322678222
0	322680306
1	322682389
0	322684474
0	322686558
1	322688641
0	322690726
0	322692813
0	322694955
0	322697073
0	322699157
[... 116 lines cut ...]
Time travel: last_triggered=1929379833us time=1895825435us travel=-33554398us
Time travel: last_triggered=1979711483us time=1946157085us travel=-33554398us
Time travel: last_triggered=2046820349us time=2013265952us travel=-33554397us
Time travel: last_triggered=2097151991us time=2063597593us travel=-33554398us
Time travel: last_triggered=2130706424us time=2097152027us travel=-33554397us
Time travel: last_triggered=2147483615us time=1us travel=-2147483614us
Time travel: last_triggered=268435448us time=234881050us travel=-33554398us
Time travel: last_triggered=369098748us time=335544351us travel=-33554397us
Time travel: last_triggered=486539261us time=452984863us travel=-33554398us
Time travel: last_triggered=553648126us time=520093728us travel=-33554398us
Time travel: last_triggered=570425340us time=536870942us travel=-33554398us
Time travel: last_triggered=620756982us time=587202584us travel=-33554398us
Time travel: last_triggered=637534203us time=603979805us travel=-33554398us
Time travel: last_triggered=654311418us time=620757020us travel=-33554398us
Time travel: last_triggered=671088636us time=637534238us travel=-33554398us
Time travel: last_triggered=738197497us time=704643099us travel=-33554398us
Time travel: last_triggered=872415225us time=838860827us travel=-33554398us
Time travel: last_triggered=889192446us time=855638049us travel=-33554397us
Time travel: last_triggered=905969654us time=872415256us travel=-33554398us
Time travel: last_triggered=973078526us time=939524128us travel=-33554398us
```
Somehow last trigger sometimes happens after current trigger.
Or current trigger happens before last trigger already happened in the past.
I do understand that 1us messages because integer overflow, but not that [b]travel=-33554398us[/b] thing.
Is this right?

pjsg, 2017-02-08:
The pin value that is supplied in the callback is the value of the pin when the interrupt handler gets to read the pin. This is a few microseconds after the interrupt is triggered. It is entirely possible that the pin value has changed.

More worrying is the backwards time travel. This suggests a bug....

Uinden, 2017-02-08:
**pjsg**,
Here i made **8.5mb** log of **642988** trigger events [trigger.txt](https://github.com/nodemcu/nodemcu-firmware/files/761264/trigger.txt)
And there is no **LOW** levels at all. So i think if it would trigger on **LOW** and then change to **HIGH** while triggering then there should be **LOW** levels in log from **642988** events but they all **HIGH**. 
I think it triggers on **HIGH**, but i may be wrong.
*Also there is time travel in log at line **556010**
```
1	1560276801
1	1560278944
1	1593835519 <---
1	1560283230
1	1560285374
```

FrankX0, 2017-02-08:
The first part of your findings is caused by the fact that you are using a digital input in it's undefined input range. According the datasheet from 0.25×VIO (~0.8V) to 0.75×VIO (2.5V).
This can (and will) cause undefined behaviour as you found out.
Furthermore you are triggering on a low level. This will cause constant triggering at high rates while the input is low. Normally triggering is useful on a falling (down), rising (up) or both edges.

Uinden, 2017-02-09:
**FrankX0**,
So **falling** and **rising** affected too.
I tested **gpio.trig** with "**up**", "**down**", "**both**", "**low**", "**high**" triggers. 
And they all produce same result. Flooding with **LOW** and **HIGH** levels depending on voltage.
I just re-tested them all to be sure and they do all flood. **down** flooding with **HIGH**, **up** flooding with **LOW**.

Initially i found this problem while learning how to use ultrasonic sensor **HC-SR04**. It has **ECHO** pin that sets **HIGH** level for some time, and that time means distance it measured. Example code was setting trigger to **up** and after triggered it was setting it to **down** and counts distance. But sometimes it was triggered with two **LOW** or two **HIGH** signals. And time between triggers is important (since it means distance) and i noticed time travel bug too.

Uinden, 2017-02-09:
I have **Arduino IDE** with **ESP8266** support installed. So i got this code:
```C
const int pin = 5;

void trigger_HIGH() {
  Serial.println("Triggered HIGH level");
  attachInterrupt(pin, trigger_LOW, FALLING);
}

void trigger_LOW() {
  Serial.println("Triggered LOW level");
  attachInterrupt(pin, trigger_HIGH, RISING);
}

void setup() {
  Serial.begin(9600);
  
  pinMode(pin, INPUT);
  attachInterrupt(pin, trigger_LOW, FALLING);
}

void loop() {

}
```
And i got flooded with
```
Triggered LOW level
Triggered HIGH level
Triggered LOW level
Triggered HIGH level
Triggered LOW level
Triggered HIGH level
Triggered LOW level
Triggered HIGH level
Triggered LOW level
Triggered HIGH level
Triggered LOW level
Triggered HIGH level
Triggered LOW level
Triggered HIGH level
Triggered LOW level
```
So now i see **NodeMCU** has nothing to do with this behavior.  
So **gpio.trig LOW triggers with HIGH level** is false issue.
Should i delete that part from my issue?

FrankX0, 2017-02-09:
Apparently you are suffering from (extensive) noise on the input.
And it seems that the ESP8266 sometimes is not capable of correctly handling the high rate of interrupts, resulting in "time travel".
In my opinion you should first create a stable environment to be able to measure the pulse width correctly.
In the mean time this issue can be closed as it currently seems not to be a NodeMCU firmware issue.

pjsg, 2017-02-09:
I've filed a PR to fix the "time travel" issue. I agree that the other issue is not an issue.

closed 2017-02-09 by marcelstoer

Uinden, 2017-02-09:
I assume PR will be accepted, so i can close issue. 
Thank you for help.

PR #1788 (2017-02-09): Fix the adjusting of the time when there is a rollover in gpio interrupts
Fixes #1782 .

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

The code that adjusted the time reported when the system time rolled over 2^24 boundary between the interrupt happening and when it was dispatched  [... cut]