nodemcu/nodemcu-firmware #1784: Cron module not working

samuel-puschacher, 2017-02-08:
Hello everybody
i want to send sensor data over MQTT every minute, so I thought it would be the best, to use the cron Module, but unfortunatly it is not working

```Lua
cron.schedule("* * * * *", function(e)
          print("Every minute")
          m:publish("/gGar/temperature","11.2",1,0, function(client) print("sent temp") end)
          m:publish("/gGar/humidity","88.3",1,0, function(client) print("sent") end)
        end)
```
### NodeMCU
My NodeMCU Version is 2.0.0
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: b9d9f5856a44515bf03cc04801e8d99b79d6bbde
	SSL: true
	modules: adc,cjson,cron,dht,file,gpio,i2c,mdns,mqtt,net,node,perf,pwm,rfswitch,rtcmem,rtctime,sntp,spi,tmr,u8g,uart,wifi,ws2812,tls
 build 	built on: 2017-02-08 10:43
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
```
### Hardware
I am using a Wemos ESP-12 on a Wemos D1 Board

What did I do Wrong?
Thanks for Help

djphoenix, 2017-02-08:
Is there any RTC time source in your code? Without RTC time, cron module should never work (as it noted in [docs](http://nodemcu.readthedocs.io/en/dev/en/modules/cron/)).

marcelstoer, 2017-02-08:
Yury, `rtctime` is listed.

djphoenix, 2017-02-08:
Marcel, I see, but is it in-sync with any source (or rtctime.set was even called)? Without sync, `gettimeofday` always returns zero, so...

marcelstoer, 2017-02-08:
Ah, that's what you mean. I see now, sorry.

off-topic
Maybe we should extend the "Do not forget to include the rtctime module." part in the cron documentation to remind users to not just include rtctime but to actually _initialise_ it properly. We could even print a warning in the cron code if `gettimeofday` returns unexpected values.

djphoenix, 2017-02-08:
Good point to note in docs, but not for warn in code. There are some situations like:
1. Device starts
2. Device loads config (with saved schedule) and starts connect to WiFi/API
3. Device connects to wifi and starts trying to sync with SNTP
4. Device synced and ready.

I think annoying messages from 2 to 4 is unnecessary...

samuel-puschacher, 2017-02-08:
Ok, thank you verry much. 
This should be added to the docu. I thought, that the clock just strarts from zero, and I would not need to set the time for the cron module.
Would it be Possible to use the cromn module for smaler intervals too, like seconds or miliseconds?

djphoenix, 2017-02-08:
Nope, `cron` module was planned as worker for long-planned repeatable tasks. For smaller intervals you may use `tmr` module (with [dynamic timers](http://nodemcu.readthedocs.io/en/dev/en/modules/tmr/#tmrcreate) introduced in `2.0.0`).

samuel-puschacher, 2017-02-08:
Are the dynamic timers also blocking the network stack?

closed 2017-02-08 by samuel-puschacher

zszen, 2017-08-30:
I set rtctime.set(0,0,0)
the cron don't work