nodemcu/nodemcu-firmware #1785: Add a more flexible PWM implementation

ARF1, 2017-02-08:
### Missing feature
The built-in PWM implementation has multiple issues:

1. Duty cycle limited to 90% (at 1kHz PWM period)
2. usable PWM period at most ~2KHz.
3. considerable jitter

It would be nice, if especially the PWM frequency could be extended upwards.

### Justification
For some applications, 2kHz PWM is insufficient. For others, the associated jitter is a problem.

### Possible solutions
I see two solutions that look interesting:

1. The [re-implementation of software PWM](https://github.com/StefanBruens/ESP8266_new_pwm) by @StefanBruens. It uses a step-size of 200ns which allows e.g. a 8-bit resolution PWM at 19kHz or 12-bit resolution PWM at 1kHz. 
2. A solution which uses the I2S peripheral for DMA-driven variable-frequency, jitter-free PWM output along the lines of https://github.com/cnlohr/esp8266ws2812i2s.

marcelstoer, 2017-02-11:
> I see two solutions that look interesting

We always welcome PRs (#1010). @pjsg you were the last to work on PWM. Maybe you can judge how well founded this is?

pjsg, 2017-02-11:
I think that you need to be very clear on the requirements and what you actually need. The I2S approach will give you one channel of very good PWM. There are other devices that use that approach -- the ws2812 drive for example.

With the regular PWM, the problem is that it is all done in software with timer interrupts. These can be delayed significantly. 200ns resolution sounds nice, but there can be many us of jitter caused by interrupt masking (it isn't actually clear to me what the maximum delay can be. I guess that it would be possible to run some experiments. 

The advantage of doing everything in software is that you can make it as complex as you like and provide different PWM frequencies on each pin. However, the interrupt handler becomes much more complex and it may not be easy to get the code right.

In short, by using the I2S approach you get good control over 1 channel (on pin 4). Using a software approach, you get control over all pins, but with jitter. The jitter is intrinsic due to the latencies between when the timer triggers and the interrupt is dispatched. The interrupt can be delayed either because interrupts are disabled, or because a higher priority interrupt is in progress.

ARF1, 2017-02-19:
@pjsg You wrote:

> The I2S approach will give you one channel of very good PWM. There are other devices that use that approach -- the ws2812 drive for example.

I am trying to find my way around the code and had a look at the ws2812 driver. From what I understand it does not use I2S but an UART. Am I misunderstanding the code?

closed 2018-02-24 by marcelstoer