nodemcu/nodemcu-firmware #1786: Bug in IIR filter settings for BME280 - incorrect pressure

mickley, 2017-02-08:
### Expected behavior
Changing the IIR filter coefficient in bme280.init() should not (greatly) affect the values returned

### Actual behavior

When the IIR filter coefficient uses setting 1 (filter coefficient of 2), the pressure readings are way off.  Instead of being in the 1000 hectopascal range (100 kPa), the returned value is around 660 hectopascals (66 kPa), far lower than any atmospheric pressure on earth.  

The filter coefficient performs averaging of samples, so with lower filter coefficients, even though more variability would be expected, nothing this extreme should happen.

Furthermore, appropriate pressure readings are returned for every other filter coefficient setting, including turning the IIR filter off completely.  

### Test code

```Lua

-- Initialize the bme280 module on pins 1 and 2, with an IIR coefficient of 2
bme280.init(1, 2, nil, nil, nil, nil, nil, 1)

-- Wait 500 ms before testing to allow the sensor to acquire multiple readings
tmr.create():alarm(500, tmr.ALARM_SINGLE, function()

    -- Get temperature, pressure, and humidity from the BME280
    temp, pressure, humid = bme280.read()

    -- Print out the pressure reading
    print(pressure)

end)

```
### NodeMCU version
dev branch, commit: a271ce4d34da4efb2a25114da7a5725d344c85ce

### Hardware
WeMos D1 Mini with 4MB flash

FrankX0, 2017-02-08:
I had a quick look at the bme280.init function.
The 8th parameter (3 bits) is shifted by 1 bit to the left to go into the config register:
```
config = ((!lua_isnumber(L, 7)?BME280_STANDBY_TIME_20_MS:(luaL_checkinteger(L, 7)&bit3))<< 4) // 7-th parameter: inactive duration in normal mode
| ((!lua_isnumber(L, 8)?BME280_FILTER_COEFF_16:(luaL_checkinteger(L, 8)&bit3)) << 1); // 8-th parameter: IIR filter
```
So bits 1, 2 and 3 are being set.
But according the BME280 datasheet the filter is set through bits 2, 3 and 4.
With a setting of 1 in your case, this would set bit 1 in the config register, which is reserved and should not be changed. This might explain the incorrect behaviour.

Can you try using 2 instead of 1 as your 8th parameter?
`bme280.init(1, 2, nil, nil, nil, nil, nil, 2)`
If I'm correct this would set the filter coefficient correctly to 2.

mickley, 2017-02-08:
@FrankX0  Yes, 2, 3, 4, and 0 all give proper values.  Harder to test whether the filter is working correctly, but that explanation makes sense to me.

Cheers
- James

PR #1787 (2017-02-11): Correcting BME280 IIR filter setting
Fixes #1786.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Parameters "inactive_duration" and "IIR_filter" were incorrectly written to the config register.
@mickley: can you test?
@vsky279: do you agree?
closed 2017-02-11 by marcelstoer