nodemcu/nodemcu-firmware #1789: Do we need / want protected call of task callbacks?

wolfgangr, 2017-02-09:
8<------------------------ BUG REPORT -----------------------------------------

If I call `lua_err` from a task callback, it raises a **panic** instead of the error handler function configured at the state where everything startet (in my test environment: the serial lua console)

### Test code
register error handler as a task callback:
`taskno_error = task_get_id(net_if_error);`

post a task with the error handler and an error message as parameter:
```
const char msg[] = "foo bar error";
task_param_t param = (task_param_t) msg;
task_post ( NET_INFO_PRIORITY_ERROR, taskno_error,  param );
```

raise a lua_error in the error handler:

```
static void   ICACHE_FLASH_ATTR  net_if_error (task_param_t param, uint8_t prio ) {
...
  lua_pushstring(L , errmsg);
  lua_error(L);
```

### Expected behavior
stack trace and return to the console
```
foo bar error
stack traceback:
	[C]: in function ' net_if_error'
	stdin:1: in main chunk
>
```

.... 
### Actual behavior
panic and reboot
```
PANIC: unprotected error in call to Lua API (foo bar error)
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

### NodeMCU version
dev
NodeMCU 2.0.0 build unspecified powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Hardware
Mini NodeMcu ... TE441
http://www.ebay.de/itm/282013844997

### Workaround
either use print
```
  lua_getfield(L, LUA_GLOBALSINDEX, "print");
  lua_pushstring(L, "net_info ERROR: ");
  lua_pushstring(L , errmsg);
  lua_call(L, 2, 0 );
```
and / or provide for an error callback in the api

8<------------------------ END BUG REPORT -------------------------------------

### Explanation:

I conclude that the configured **error handler** from the console environment is **not available** in the task callback context. Therefore,` lua_error` falls back to the **default behaviour** which is to **panic**.

https://nodemcu.readthedocs.io/en/master/en/lua-developer-faq/

> It is important to understand that any **event callback task is** associated with a single Lua function. This function is **executed** from the relevant NodeMCU library C code **using a** ` lua_call()`.
	.... No locals are retained across events..... 

To avoid this panic, I think we'd need the task callback invoked by [lua_pcall](https://www.lua.org/manual/5.1/manual.html#lua_pcall) instead. 

I'm reluctant to advocate for this and propose the workaround instead.

wolfgangr, 2017-02-09:
I encountered the issue in my quest to get PR #1757 running.

Exemplary context: 
_What to do when the host name resolution fails in preparation of a network "ping"?_

https://github.com/nodemcu/nodemcu-firmware/pull/1757#discussion_r97462147
@pjsg wrote:
> Also, it would be better to throw a LUA error rather than just return silently.

https://github.com/nodemcu/nodemcu-firmware/pull/1757#discussion_r97626135
@devsaurus wrote:
> I'd even favor @pjsg's proposal to throw a Lua error instead, but this function is executed in some lwip-callback context. Guess the context issue has to be clarified first before.

https://github.com/nodemcu/nodemcu-firmware/pull/1757#discussion_r97461970
@pjsg wrote:
> It isn't clear to me what context this is executing in. I suspect that it might as a direct callback from lwip. In that case, you mustn't reference any lua context. You have to use the task api to get the data queued down to the base level.

I generally agree with the cited comments and tried to implement an generic error handler as a task callback. This yielded in the described behaviour.

I didn't get response to my comments in the PR discussion, presumably because this is clobbered with lots of beginerrs errors. So I'd dare to focus the core developpers attention by distilling the essence and raising this as a seperate issue.

----

My solution in the example at hand would be as follows:
- seperate the hostname resolution from the ping ( aka KISS)
- API: something like `net_info.host_to_ip("my.host.net", errorfunc)`

If the error function is omitted/nil, then a simple print ("net_info ERROR: foo bar message") is executed.
If an error function is given, it will receive the error string as its first argument on the stack.

I'd really get the ping case done and start with my ethernet driver. But the question of a generic error handler in any network related api is a common one that extends beyond ping. So I would be glad to have the issue settled instead of rewriting anything from scratch again.

wolfgangr, 2017-02-09:
Just another thought on the question of `NODE_ERR()` vs `c_printf()`. its indirect cousins, and the question of **redirection**, as raised in #1757:

When we call lua print as in my workaround section above, we shift the responsibility and possibility to redirect print output from the compiler direcitves to lua land - and I think this is where it belongs. If nobody is listening at the serial line, the best way to redirect is by redefinig `_G.print`, isn't it?
Any objections?

pjsg, 2017-02-11:
My feeling is that it is up to the programmer to handle errors. The *vast* majority of the time, there is nobody looking at the console, so printing stuff there is not "error handling". 

The current approach is to panic on errors being thrown (except from the console interpreter). This means that, if your application fails, it gets restarted automatically.

If you want to wrap your code in a `pcall` then you can do that. But think about what you want to do in the error handler. In many cases, writing the recovery code is more complex than the original code. In most cases, it is easiest just to restart the platform.

wolfgangr, 2017-02-11:
@pjsg :
> My feeling is that it is up to the programmer to handle errors.

Agree - at least when the application is ready for deployment.
In early development stage, however, the programmer will be happy to get a hint on the console.

> The vast majority of the time, there is nobody looking at the console, so printing stuff there is not "error handling".

That depends on the usage.
I'd like to deploy nodeMCU in sensor applications of small lot size (quite often 1)
So I expect a development stage and a use stage. I'd like to switch over as simple as possible.
My proposal was with this scenario in mind.

> If you want to wrap your code in a pcall then you can do that. 

How can I do that on a task callback? I am not the caller, there.
The task callback is called by the task queue manager of the os.

So, raising an lua error in a task callback would require somehow to tell the os what my intended error handler were.

>  In many cases, writing the recovery code is more complex than the original code.

Agree. 
The point is that there are situations which are expectable errors.
In the ping example where this case came up:
- try to resolve a hostname with a typo
- try to resole a hostname with wifi down

This is what the programmer wants to handle with his own routines.
The "lua_error" mechanism may be one way to do so, a quite generic one, but not the only way.
Obviously it is not easily available in task callbacks.

So instead of making complicated exercises to make it work in task callback situation, i'd propose a workaround.

wolfgangr, 2017-02-12:
@wolfgangr wrote:
> To avoid this panic, I think we'd need the task callback invoked by lua_pcall instead.

Sorry for this sentence .
This is bullshitt, of course, as both `lua_pcall` and `lua_call` refer to call from C-land to lua-land.

Therefore, 
> event callback task is associated with a single Lua function. This function is executed from the relevant NodeMCU library C code using a `lua_call().`

does **not resemble the cause** of my problem, but presumably **shares the same cause**:
**The error function as configured for the console ist lost.**

This however were required for a proper [lua_pcall](https://www.lua.org/manual/5.1/manual.html#lua_pcall)
```
lua_pcall
          int lua_pcall (lua_State *L, int nargs, int nresults, int errfunc);
```
as well as for a [lua_error](https://www.lua.org/manual/5.1/manual.html#lua_error)

@pjsg 
> My feeling is that it is up to the programmer to handle errors. 

Perfectly agree with that. So, writing a module, we try to provide both no-surprise defaults and a maximum of flexibility to the programmer using the module.
I think **panic** is a bad surprise for a callback your are triggering from a state with properly configured error handler - be it the default of the console or your own one in some full fledged app.

Could we use [lua_atpanic](https://www.lua.org/manual/5.1/manual.html#lua_atpanic)
```
lua_atpanic
          lua_CFunction lua_atpanic (lua_State *L, lua_CFunction panicf);
    Sets a new panic function and returns the old one. 
```
to **retrieve** the error function before we enter the **quagmires of lwip and task-api** and **restore it** , before we come back to lua-land at the end of the task call back. I think this is not only the expected default behaviour for the error handler, but also for other user supplied lua callbacks. They too should not panic in any case, but display the error function as before.

Is there any reason why we shold / can / must not do this?
Can we rely / hope that all the context required  is still at hand? 
Why is it that the designer of the task api "forgot" the error handler?
By design? By necessity? Or by lack of necessity?

I'll go and try.

pjsg, 2017-02-12:
How about something like:

```
function wrap(x)
  return function(...) 
    local args = {...}
    local status, err = pcall(function() x(unpack(args)) end)
    if not status then
      print(err)
    end
  end 
end

tmr.alarm(0, 1000, 0, wrap(function() do something one second later end))
```

i.e. `wrap` is function that take a function and returns a function that takes the same arguments but just prints the error message (if any). The returned function doesn't return anything (this is OK as callbacks typically don't return anything).

Once you are ready for production, you can either remove all the wrap calls, or change `wrap` as

```
function wrap(f)
  return f
end
```

wolfgangr, 2017-02-12:
I think we still are missing each others point.

Let's take the **example** I'm just working on: 
**DNS hostname resolution** ( PR #1757)
I separated it out of the ping to reduce complexity and increase modularity.
I use lwip instead of espconn now.
So, following your advice, I don't ask for a lua state in the lwip callback, but insert an extra task callback stage.

DNS name lookup typically is a **blocking operation.** 
**Typical errors** are **"host not found"** or **"DNS not available"**. 
These error states are typically available after some **timeout**.
This are typical examples of **runtime errors**, which are **not restricted to the development stage** of an application.

Now take a **typical application** that people have already announced:
NodeMCU as a **network surveillance gadget**, performing pings / DNS lookups and whatever on a repetitive scheme and **reporting the result** via mqtt, mail, blinking LED or whatever.

The idea now was to raise a lua_error in the task callback, which the application programmer could use to hook his specific action in. 
**It's no help, when the module is rebooting** each time it's supposed to do its intended job.

I can't see how the lua wrapper approach could help, too.
What shall I wrap? The caller for `dns_gethostbyname`?
This does not help and does not change anything. First, because in this caller, the error callback is still valid, so I could rise lua_error without any problem. Second, the DNS error state is not yet available in the caller, but earliest in the lwip callback.

But in the lwip callback you told me not to ask for a `lua_getstate`. I'm still courious why. I can guess that lwip callbacks could originate from interrupt handlers? anyway...

Obedient as I am, I post **task callbacks**, where anything else is fine, but **the error handler** from the caller **is gone.** I tried to save and restore it using `lua_atpanic`, but it looks like atpanic is not the error handler referred to by `pcall`.

I think I'll scrap the idea of throwing lua_error and return just some fields to the final lua callback where the application programmer can retrieve the status of DNS lookup (or whatever else). In my opinion, the goal should be to keep the api as thin, flexible and simple as possible. In the current case, throwing errors spoils these goals in favour of some kind of questionable conceptual beauty.

pjsg, 2017-02-12:
Lets take the example of looking up a DNS name...

```
net.dns.resolve("www.google.com", function(sk, ip)
    if (ip == nil) then print("DNS fail!") else print(ip) end
end)
```

In this case, there is a single callback that is always invoked. Either the lookup works and we have an IP and it is passed in, or the lookup fails and there is no IP. In both cases, the callback is invoked. I guess that there is a third case where something catastrophic goes wrong (e.g. running out of memory) and the platform panics.

Another style is the `mqtt.client:connect()` approach where there are two callbacks provided. One is invoked in the case of success, and the other in the case of error. Exactly one of the callbacks is (supposed to be) invoked.

A third style is when there is an `on()` method that attaches callbacks to the object. You can go:

```
thing = something.create()
thing:on('error', function() print ('error') end)
thing:on('success', function() print ('success') end)

thing:startup()
```

These three styles are all found in nodemcu. I'm sure that different people have different views on which one is best. 

In the case of a `ping` method, I'd favor the single callback approach as it least likely to have programmer error.

```
ping(address, [ options, ] callback)
```

Where `address` is an address or a name, `options` is a table with options (e.g. `count`, `size`, `ttl`, `timeout`), and `callback` is a function that is guaranteed to be invoked exactly once.

`ping` may also throw an error if there is an existing ping running that has not (yet) completed. It would probably also throw if the arguments were invalid in some way (e.g. no arguments provided). 

`callback` is invoked with two arguments -- `result` which is a table that summarises the ping responses, and `error` which is a string describing the error (if any). The `result` table might contain `sent`, `received` (counts of packets sent/received), `rtt-min`, `rtt-max`, `rtt-avg` which are the round trip times in milliseconds. Could also include information about the fate of each ping packet -- you may want to report on 'icmp unreachable' etc.

wolfgangr, 2017-02-12:
So we can **finally agree** that for runtime errors, throwing an lua_error may not the best suitable strategy :-)
Many thanks for the elaborated explanation of api design, "how it is done around here"

I don't agree with all detailed conclusions regarding the ping. I think there is a staged event cascade in a typical console ping: every second, an request ist sent out and the response is protocolled. At the end, a summary is presented. 
I'd allow three different callbacks:
- one for th individual round trips
- one for the summary
- one for printing - so we can either redirect or suppress printing as an easy alternative to callbacks

But this not the case of the issue at hand. For this, the essence is that in no case we require catchable lua_error for run time errors.

Agree?

wolfgangr, 2017-02-12:
@pjsg 
I see that your [wrapper approach](https://github.com/nodemcu/nodemcu-firmware/issues/1789#issuecomment-279225510) is of course a proper **remedy for the second problem** raising from the issue at hand:

When the error handler is lost during the callback quagmire, any **user supplied callback that is executed from this** at the end hits the hard **panic** for any error, instead of a graceful message.

So I think I'll put a hint in the **documentation**. May I include your wrapper in verbatim? Or is it somewhere in the docu already? or is it OK to point from the docu into the issue list here?

pjsg, 2017-02-12:
Just copy the wrapper (after making sure that it works). I did test it on my linux lua 5.1, but not on nodemcu!

wolfgangr, 2017-02-12:
Thank you :-)
From my point, I'm happy.

Let me recapitulate:
- during the **task api,** any preconfigured **error handler** from the registering environement **gets lost**
- any lua error therefore triggers the **default** error handler, which is to **panic**
- we should **not call** `lua_error` from C task callback handlers
- to **protect user** supplied lua task **callbacks** we can use @pjsg [wrapper function](https://github.com/nodemcu/nodemcu-firmware/issues/1789#issuecomment-279225510)

I think we might close the issue now.

closed 2017-02-14 by marcelstoer