nodemcu/nodemcu-firmware #1791: weird gpio action

zenz, 2017-02-10:
NodeMCU custom build by frightanic.com
	branch: master
	commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
	SSL: false
	modules: dht,file,gpio,mqtt,net,node,tmr,uart,wifi
 build 	built on: 2017-02-09 08:15
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

I'm trying to use nodemcu to control a relay. if I do gpio.write(1, gpio.HIGH), then the relay is actived. but when I do gpio.write(1, gpio.LOW), the relay is still actived. until I do gpio.mode(1, gpio.INPUT), the relay goes deactived. Please test

NicolSpies, 2017-02-10:
I assume you drive the relay via a transistor and with a flyback diode across the relay coil

pjsg, 2017-02-11:
What is your drive circuit?

zenz, 2017-02-14:
@NicolSpies is right, while I replace with different relay module, it's working fine.

zenz, 2017-02-14:
Please close this issue.

closed 2017-02-14 by djphoenix