nodemcu/nodemcu-firmware #1792: Improve the MQTT example in the documentation

marcelstoer, 2017-02-11:
The one-size-fits-all example at https://github.com/nodemcu/nodemcu-firmware/blob/master/docs/en/modules/mqtt.md#example is not ideal IMHO. Particularly, shouldn't the `publish`/`subscribe` functions only be invoked when you're certain that the client is connected (i.e. in the connect-callback)?

@NicolSpies and @pjsg what's your take on this?

NicolSpies, 2017-02-11:
@marcelstoer You are correct. 
The example appears to follow the sequence how the client should be used but it can be better described as a sequential listing of examples of the syntax of the methods and events that loosely follows the correct calling sequence.
I think the individual methods and events should all be augmented by  "single line" examples like already implemented and then a more representative working example in line with what you suggested.

marcelstoer, 2017-02-11:
Thank you Nicol. I'm not entirely sure I got your point, though. Could add a proposal here (pseudo code would be enough)?

I feel the client functions are well documented. A documentation can always be improved, of course, but they're complete and not misleading.
The initial example is just wrong because it ignores that event-driven nature of NodeMCU; running it will fail. For now I'd only move the `publish`/`subscribe` calls in that example.

NicolSpies, 2017-02-11:
I agree, with all the possible combinations of using the methods and events an example of use will indicate a single case of application, better leave it out.

commit b26ed972 (2017-02-12): Improve MQTT client example, fixes #1792
closed 2017-02-12 by marcelstoer