nodemcu/nodemcu-firmware #1795: DS18B20 module does not approach correctly the delay needed for temperature conversion

vsky279, 2017-02-12:
### Expected behavior
Before reading DS18B20 sensor the reader needs to wait before the conversion is finished. This applies especially to parasit-powered sensor where I think the scratchpad will not be holding last reading (so no temperature is returned).

### Actual behavior
In the existing example this is "solved" by the ([commented](https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/ds18b20/ds18b20.lua#L86)) `tmr.delay(1000000)`. This is not the right approach in the NodeMCU Lua. Waiting 750 ms, the minimum delay before conversion finishes, makes the chip unstable.
This becomes even more unbearable if more sensors are connected and more delays need to be called. Only one conversion can occur at time for a parasitic-powered sensors due to electric current restriction.

### NodeMCU version
Latest dev branch

### Hardware
Any

### Proposed solution
A callback approach should be implemented.
I propose the following implementation [ds18b20-async.lua](
https://github.com/vsky279/nodemcu-firmware/blob/DS18B20/lua_modules/ds18b20/ds18b20-async.lua).
I would even suggest that the previous example is removed as I think it is misleading.

*Issue:* my understanding was that I can start more conversions at time for powered sensors (non-parasitic) but this does not work correctly for me. I need to start conversions [one by one](https://github.com/vsky279/nodemcu-firmware/blob/DS18B20/lua_modules/ds18b20/ds18b20-async.lua#L17). In my hw setup several powered sensors are connected + 1 parasitic-powered.

### Example
```lua
t=require("ds18b20-async")

function readout(temp)
  local resp=""
  for addr, temp in pairs(temp) do
    -- print(addr, temp) -- better representation base64 is preferred (when encoder module is included)
    print(encoder.toBase64(addr), temp)
  end
end

t:readTemp(readout)
```

eyaleb, 2017-02-12:
@vsky279 "unbearable" you say. Agreed, so I never wait for a conversion.

What I do is:
- read data
- start conversion
- report data
- sleep [the conversion completes in the background]
- repeat

The server receiving the data can handle the fact that each report belongs to the earlier period, if it matters.

Regardless, a fixed 750ms delay is not required, one can check the status following the `CONVERT T` until it is done. I measured a wait of 660ms on average. YMMV.

vsky279, 2017-02-12:
@eyaleb If you're reporting data every few seconds then why not. But would it be the best approach if you want to deep sleeping between? 
Would it work correctly for parasitic-powered sensors or mix of both?

Moreover the example is not doing what you are suggesting as it starts conversion and then reads data immediately which is not acceptable for parasit-powered. The [documentation](http://datasheets.maximintegrated.com/en/ds/DS18B20.pdf) says: For parasite-powered DS18B20s, the master must enable a strong pullup on the 1-Wire bus during temperature conversions and copies from the scratchpad to EEPROM. **No other bus activity may take place during this time.**

The module might be a bit misleading for beginners who might be tempted to uncomment the `tmr.sleep()` line.

What I suggest is to:
- start converstion on sensor Nr. 1
- wait using timer for the conversion to complete
- read data on Nr. 1
- start converstion on sensor Nr. 2
- wait using timer for the conversion to complete
- read data on Nr. 2
- ....
- report data in callback routine
- repeat

I also tried to (but w/o success):
- start converstion on sensor Nr. 1, Nr. 2, ... (all powered sensors) and on the first parasit-powered Nr. P
- wait using timer for the conversion to complete
- read data on Nr. 1, Nr. 2, .., Nr. P
- start converstion on sensor Nr. P+1 (parasit-powered)
- wait using timer for the conversion to complete
- read data on Nr. P+1
- ....
- report data in callback routine
- repeat

karrots, 2017-02-12:
@vsky279 are your parasitic and non-parasitic sensors on the same bus? Do you individually address the sensors or use the broadcast functionality to trigger the temperature conversion.

vsky279, 2017-02-12:
@karrots Yes, I'm mixing powered and parasit-powered sensors on the same bus. In my implementation I have several powered and 1 parasit-powered sensors. According to http://www.best-microcontroller-projects.com/ds18b20.html it is allowed.

I need to address sensor 1 by 1. If there were more parasit-powered sensors I'm afraid the bus would not supply sufficient voltage (through the resistor) to run two conversions at once.

It is nicely discussed here: https://www.maximintegrated.com/en/app-notes/index.mvp/id/4255.

karrots, 2017-02-13:
Why do you need to cycle through each instead of just issuing `skip rom` (CCh) to trigger the conversion for everyone? Then read each individually later? I have done this with success on another project (not NodeMCU based). I need to do it on a future project (NodeMCU based) but haven't made it to that part of the coding.

When attempting to start all the conversions then read all of the temps did all sensors fail equally or just one type (parasitic vs powered)? Have you verified things look with a logic analyzer? What is your power source for the DS18B20s?

vsky279, 2017-02-13:
@karrots If you have more than 1 parasit-powered sensor on the bus you should not start conversion on both at the same time. The current consumption would be too high and the voltage would drop below the required minimum level causing sensors to reset.

Usually all sensors failed when the parasit-powered sensor was connected. If it was not connected there was no problem to start conversion for all sensors. The safe way was to start conversion one by one. Unfortunately my device is not accessible easily so I didn't really study why starting conversion on all sensors failed.

karrots, 2017-02-14:
We had issues with parasitic code in OW in the past I was trying to get a feel for if the issue was code related or not. 

I assume the non-parasitic sensors are powered directly from Vdd and not some other pin.

vsky279, 2017-02-14:
Yes, non-parasitic are powered directly from V<sub>dd</sub>.

devsaurus, 2017-02-19:
> I would even suggest that the previous example is removed

@vsky279 I agree. The existing example isn't working for me with a parasite powered sensor due to the `tmr.delay()` issue. Your script shows how to address this with callbacks.
Note that temperature calculation only works for an integer build, too many decimal points for a float build 😄 

I also checked with 2 sensors on the bus: a V<sub>dd</sub>-powered and a parasite one. Had to tweak [the script a bit](https://gist.github.com/devsaurus/a11cf1d5c297a37a7835b061d67305f1), but parallel conversion works for my configuration now.

devsaurus, 2017-02-19:
The `conversion()` method does only a single `ow.reset()` for the first sensor in the list. This seems to break the conversion for subsequent sensors (sometimes). Datasheet is quite strict about this in chapter "Transaction Sequence".
I've updated my gist accordingly.

vsky279, 2017-02-19:
@devsaurus That's a good point. I will need to test it. Seems that it might be the cause for my unability to do parallel reading for powered sensors.
I have added part for float version. I don't think it make sens to make the module universal.

vsky279, 2017-02-19:
@devsaurus Perfect! With this fix it works as it should work, i.e. conversion starts on all powered + first parasite powered and all readouts performs well.
I will prepare a PR with an example to be used with this module and removing the old one.

PR #1820 (2017-02-25): DS18B20 module (asynchronous)
Fixes #1795.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Implements callback approach to DS18B20 sensors readouts. Eliminates the `tmr.delay()` approach in previous version of the module. See #1795 for details.
closed 2017-02-25 by marcelstoer