nodemcu/nodemcu-firmware #1798: Another Lua for ESP???

joysfera, 2017-02-13:
### another Lua for ESP?

I have just found this and I'm so surprised by its existence (given the fact that I'm waiting for NodeMCU for ESP32 impatiently since last summer and it seems like this summer it's still not going to be ready) that I'm opening an issue as I don't know any other reasonable way how to let you all know that the NodeMCU (not only for ESP32) might perhaps somehow benefit from this: https://github.com/loboris/lua-rtos-esp32-lobo

Feel free to delete it if it's a common knowledge but I myself am really shocked. I always thought that NodeMCU was the only Lua for ESP.

joysfera, 2017-02-13:
I went ahead and created an issue there as well, just to see what's up: https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/19

marcelstoer, 2017-02-13:
We've had an [RTOS branch](https://github.com/nodemcu/nodemcu-firmware/tree/dev-rtos) since last May. However, ever since Espressif made more or less clear that the only way forward is [ESP-IDF](https://github.com/espressif/esp-idf) (at least that's my understanding) that became stale.

joysfera, 2017-02-13:
Well, the whitecatboard seems to be based on esp-idf. And seems to work (judging from the changes made by loboris - mostly adding drivers for real HW). Pity they don't use the event driven programming model that I love in NodeMCU.
Anyway, can NodeMCU borrow something from whitecat to boost the current ESP32 branch that slowly builds on top of esp-idf? ESP32 is no longer a mythical chip, it's being sold almost everywhere now and its support is becoming essential.

the0ne, 2017-02-14:
Not sure how up-to-date their website is but the functionality matrix at the bottom of http://whitecatboard.org/software/lua-rtos/ lists a lot of basic features like ADC, SPI, UART, etc. as "planned".

joysfera, 2017-02-14:
I have asked them in the issue listed above and they basically told me to f-off, if I can read their reply properly, so I believe this issue can be closed.
As for features - since loboris wrote functional drivers for advanced hardware like TFT displays I believe it's working well enough to get some stuff running. Haven't tried it yet, still believing in nodemcu firmware :)

devsaurus, 2017-02-14:
> I always thought that NodeMCU was the only Lua for ESP.

There's [NodeLua](https://github.com/haroldmars/nodelua) on esp8266 and [LuaNode](https://github.com/Nicholas3388/LuaNode) for both the esp8266 and esp32. Never tried any of these.

> Anyway, can NodeMCU borrow something from whitecat to boost the current ESP32 branch that slowly builds on top of esp-idf?

I think it's more a lack of time than technical issues to extend our ESP32 branch. @jmattsson did a great job to get it running, but the beast hasn't gained enough momentum yet. It would need more contributors to bring it on the next level. Johny laid out the path quite well at https://github.com/nodemcu/nodemcu-firmware/projects/1.

joysfera, 2017-02-14:
NodeLua is dead, LuaNode is alive (last commit yesterday).
So there are three different work-in-progress projects for Lua based on esp-idf. Hmm.
Maybe @Nicholas3388 is more inclined to cooperate that the guys from WhiteCat?

jolivepetrus, 2017-02-16:
Hi @joysfera

I do not understand your affirmations about our intentions to cooperate. Please read our response to your issue published in our repository (https://github.com/whitecatboard/Lua-RTOS-ESP32/issues/19).

I believe that Whitecat and NodeMCU try to solve different problems. In our case the goal is to link the educational world with the professional world. For that reason we start our project from the scratch. Both projects have their own timings, budgets, goals, etc.

The Whitecat source code is available to the community under the BSD license, and anyone can do with it whatever they want.

I want to apologize to the NodeMCU team for this post in their repository, but I had to respond to statements that do not fit the reality.

joysfera, 2017-02-16:
@jolivepetrus thanks for your explanation and for your open source implementation of Lua 5.3.2 for ESP32. Unfortunately, as the API is significantly different I am afraid it will not be possible to write Lua applications that would run on both Lua implementations without major modifications. This is not a real problem, of course, just an unfortunate divide in the Lua on ESP32 world.
By cooperation I meant working towards a single API, ideally the NodeMCU one :). Your polite answer about different goals and problems shows that any unification is not likely so that's why I personally turned my attention back to NodeMCU (as I have invested two years into developing applications on NodeMCU so I am not willing to start rewriting them for a different API).
Wish you best luck with your project and I'm looking forward to giving it a try when you release some ready-to-burn binaries.

closed 2017-02-21 by marcelstoer