nodemcu/nodemcu-firmware #1799: WS2812 Module - Random colour triggered on bootup/restart

georeb, 2017-02-14:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

### Expected behaviour
WS2812 LED to remain off (0,0,0) during reboot/start up.

### Actual behaviour
On boot up (power on) or soft reset, garbage data is produced on GPIO2 which causes the WS2812 LED to display a random colour.

This can be overridden by issuing
```Lua
ws2812.init() 
```
followed by
```Lua
ws2812.write(string.char(0, 0, 0)) 
```
but there is always a delay.

### NodeMCU version
Master branch - 2.0.0

### Hardware
ESP8266-12F with a single WS2812 LED connected on GPIO2.
The data connection has an inline 0.1K resistor.

Alkorin, 2017-02-14:
Not sure we can do anything for it. I'll take a look but I think that this "garbage" is outputted by the bootloader :/

georeb, 2017-02-14:
@Alkorin Yes, I think so. It is also the pin connected to the blue indication LED. Do you know of any electrical tricks that work, for example, holding the data line low??

Alkorin, 2017-02-14:
Yes, you could use another pin of your ESP8266 to controle some kind of buffering, like the 74HCT245 which is often used with WS2812 to boost the data line to 5V. With the `OE` pin you can control if the data can go through or not.

georeb, 2017-02-14:
Okay, fair enough, it's a shame the bitstream cannot be generated on a non-uart gpio :(
It appears to only happen occasionally on a soft-restart, and is always green!

Alkorin, 2017-02-14:
It's a trade-off. I use the UART to generate de bitstream (leaving the CPU doing whatever it needs). Else you should do bit-banging and use 100% resources of the CPU because WS2812 protocol is very timing-critical...

Personally, I control hundreds of WS2812-like LEDs via Wifi at ~100fps without any crash, which wasn't really the case when the code used a bit-banging style version ;-)

georeb, 2017-02-14:
Yes, I understand. It's a small price to pay for less CPU usage :)

devsaurus, 2017-02-14:
>  think that this "garbage" is outputted by the bootloader

Appears to be correct - there's a related statement in the [esptool.py wiki](https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection#required-pins):

> GPIO2 should always use a pullup resistor to VCC, not a direct connection. This is because it is configured as an output by the boot ROM.

And in https://github.com/esp8266/esp8266-wiki/wiki/Pin-definition:

> GPIO2 	UART Tx during flash progamming

Would be interesting to see what the bootloader sends on GPIO2...

FrankX0, 2017-02-16:
> Would be interesting to see what the bootloader sends on GPIO2...

Well, this (at 74880 bps):
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26892, room 16 
tail 12
chksum 0x79
ho 0 tail 12 room 4
load 0x3ffe8000, len 2228, room 12 
tail 8
chksum 0x27
load 0x3ffe88b4, len 136, room 0 
tail 8
chksum 0xd0
csum 0xd0
```

georeb, 2017-02-19:
@devsaurus What value pull-up resistor would you suggest?

devsaurus, 2017-02-20:
@georeb the [NodeMCU Devkit](https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/NODEMCU_DEVKIT_V1.0.PDF) uses 12k for the pull-ups.

georeb, 2017-02-21:
Unfortunately the 12k pull-up doesn't fix the problem :(

georeb, 2017-02-21:
Is there any way to disable the bootloader spool on GPIO2 before build @marcelstoer ??

devsaurus, 2017-02-22:
@georeb no pull-up will fix the problem of course. I just quoted the statement from the wiki to have another source that confirms that UART is outputting information on GPIO2.

Alkorin, 2017-02-22:
If it's the bootloader, in ROM, nothing can be done ... :/

Put your code in init.lua to disable the leds as soon as possible.

closed 2017-02-23 by marcelstoer