nodemcu/nodemcu-firmware #1800: uart freez after write command

Wifsimster, 2017-02-14:
### Expected behavior
uart.write() don't freez the device

### Actual behavior
uart.write() freez the device

### Test code
I tried to write with uart when i receive a MQTT message, but the uart.write() function get stuck after execution, only a hard reboot work.

```Lua
m:on("message", function(m, topic, data)
    if data then
        print(data)
        if data == "ON" then
            uart.setup(0, 9600, 8, 0, 1)
            uart.write(0, 0xA0, 0x01, 0x01, 0xA2)          
        elseif data == "OFF" then
            uart.setup(0, 9600, 8, 0, 1)
            uart.write(0, 0xA0, 0x01, 0x00, 0xA1)
    end
end)
```

I hope, i'm just doing something wrong with the uart module...

### NodeMCU version
I used the last version of the master branch from the NodeMCU custom build.

### Hardware
I used an esp-01.

georeb, 2017-02-14:
You are missing an 'end' from your code. Try this...

```
m:on("message", function(m, topic, data)
    if data then
        print(data)
        if data == "ON" then
            uart.setup(0, 9600, 8, 0, 1)
            uart.write(0, 0xA0, 0x01, 0x01, 0xA2)          
        elseif data == "OFF" then
            uart.setup(0, 9600, 8, 0, 1)
            uart.write(0, 0xA0, 0x01, 0x00, 0xA1)
        end
    end
end)
```

Wifsimster, 2017-02-16:
Sry, just a typo i'm actually using the missing 'end' :/

Wifsimster, 2017-02-16:
@marcelstoer Could you re-open the issue ?

djphoenix, 2017-02-16:
How do you checked that UART actually stuck? I mean with logic analyser, execution flow, or just console monitor from PC?

Wifsimster, 2017-02-16:
- The relay change state at the first time MQTT message
- I send a new message (ON or OFF) and i have no print in the console (ESPlorer) and the relay don't move

I have tracked the problem with log in the console, and without uart command everything is ok.

You see nothing whrong with my setup() and write() declaration ?

devsaurus, 2017-02-16:
Does it freeze if you only remove the `uart.setup()` call and keep `uart.write()`?

FrankX0, 2017-02-16:
I cannot reproduce what you are seeing.
Both the `setup` and `write` i.c.w. receiving an MQTT message does not freeze the device.
The `setup` is correctly changing the baudrate, the `write` produces the characters one would expect.

Wifsimster, 2017-02-17:
@devsaurus If i remove the setup(), it's not stuck anymore by i see some wierd characters in the console (output of the write command without a print() ?)
```
ON
 ¢    -- Output of uart.write(0, 0xA0, 0x01, 0x01, 0xA2)
OFF
        -- Output of uart.write(0, 0xA0, 0x01, 0x00, 0xA1)
```

devsaurus, 2017-02-17:
`uart.setup(0, 9600, 8, 0, 1)` is not 100% conforming to the [API doc](http://nodemcu.readthedocs.io/en/dev/en/modules/uart/#uartsetup). Parameter `echo` is missing, but it appears to be optional anyway. Maybe try with an additional 1 for echo.

marcelstoer, 2017-02-17:
> Parameter `echo` is missing, but it appears to be optional anyway.

My interpretation of the documentation is that it's _not_ optional. The parameter description isn't very clear but the syntax shows it as mandatory. The code, however, tells a different story: https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/uart.c#L94

Wifsimster, 2017-02-17:
If my memory serves me right, i already tried with the optional `echo` and the relay didn't respond.
I'll try again tonight.

devsaurus, 2017-02-17:
Apart from the echo thing, I'd suggest that your script calls the `uart.setup()` command only once during initial configuration and not for each mqtt message.
I'm not 100% sure here, but the UART might get into trouble if it's reconfigured while sending characters from `print(data)`.

zelll, 2017-02-17:
@Wifsimster Are you trying to drive a relay via TXD of ESP-01, while TXD is also connected to your PC? 
Something you can try:
1. Select 9600 baudrate in ESPlorer.
2. Change 0xA0, 0x01, 0x00, ... to some visible characters when you debug in ESPlorer.
3. If RXD is connected to relay, disconnect it.

Wifsimster, 2017-02-17:
@zelll This is exactly that, yeah.
Disconnecting RXD seems to fix the problem :+1:  (some explications will be great :))
I don't have change the bytes in the write command

closed 2017-02-17 by Wifsimster