nodemcu/nodemcu-firmware #1802: Device rebooting with firmware with tls module

baymer, 2017-02-15:
I have nodemcu v3 board.
I build firmware on [online](https://nodemcu-build.com/index.php).
If I add tls to the build, after flashing to the board, it rebooting in loop.
And I see these messages in the serial:
```
load 0x40100000, len 30396, room 16 
tail 12
chksum 0xc1
ho 0 tail 12 room 4
load 0x3ffe8000, len 2428, room 12 
tail 0
chksum 0xb3
load 0x3ffe897c, len 136, room 8 
tail 0
chksum 0xba
csum 0xba
sy�.WV��Ʌ��error, use last saved param!
rf_cal[0] !=0x05,is 0x04

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

sandeeptara, 2017-02-15:
Same issue I am facing currently , I read somewhere that this comes if files listed below  (specially init data) are of old version (or you have to erase all the flash before reload)
0x0000 "boot_v1.6.bin
0x1000 espruino_esp8266_user1.bin 
0x3FC000 esp_init_data_default.bin
0x3FE000 blank.bin

djphoenix, 2017-02-16:
@baymer you have not-updated (but valid) RF calibration data at `0x3fc000`. Please read [flash docs](http://nodemcu.readthedocs.io/en/dev/en/flash/).

@sandeeptara can you provide code (full `init.lua` file) that you're use? I see your program is failed to bind some port, so I think there is not `tls`-related issue.

marcelstoer, 2017-02-16:
See #1010 for support.

closed 2017-02-16 by marcelstoer

baymer, 2017-02-16:
@djphoenix, I've tried to erase_flash, to flash esp_init_data_default.bin to 0x3fc000, but it's not resolved the problem. If I flash any firmware without tls, it's successful working, if with tls, it's blinking.