nodemcu/nodemcu-firmware #1807: Nodemcu Unofficial FAQ

gbechtinger, 2017-02-18:
I have something to say that maybe sounds a little offensive, so forgive me.

There a info within the FAQ that really makes me think of abandon the NodeMCU at the start, is this:

> The simplest and safest way to use IoT devices is to control them through a dedicated general purpose system on the same network. This could be a low cost system such as a RaspberryPi (RPi) server

> If you are trying to implement a user-interface or HTTP webserver in your ESP8266 then you are really abusing its intended purpose. When it comes to scoping your ESP8266 applications, the adage Keep It Simple Stupid truly applies.

I guess that with the use of "volatile modules" the NodeMCU can be easily replace the Raspberry as a central server for IOT.

Searching on google we found millions of pages talking about "No Memory" error, and this really scares the begginer. I just want to tell for those who having that type of problem that you are doing one of the following things:

- Your code is wrong
- You are not using "volatile modules"

I have a program with more than 500 lines, doing web request every 2 seconds to a web server, with an http user-interface that comunicates with the user browser every 1 second and can hold multiples users at the same time and talks and receive messages through UDP. With all that my NodeMCU is with 22000 free heap stable.

NicolSpies, 2017-02-19:
No offence taken. I believe the message is that all platforms have limitations and an optimal application fit. If that is not coming across, you are right that the message should be improved. 

There is a vast distributed knowledge and experience base in all of us.
I personally would be grateful to learn from your experience of maximizing the limited NodeMCU resources. You do not have to explain what you are doing, a representative example is all I need to figure it out and do a nice write up for everyone's benefit.

gbechtinger, 2017-02-19:
Sure, here is some of the tricks that I'm doing:

> When I need to send pages of the web interface to the clients, after receiving the request, the referenced file is readed from the file system and sent to the client. There is no HTML or other resourcess hardcoded or in memory.

> Send from the nodemcu only the the HTML page and leave the STYLE.CSS and SCRIPT.JS and images to be loaded by the client from some cloud, e.g, Dropbox

> Use volatile function for all the things that you do less, at least, and don't forget the "package.loaded[module]=nil" line at the start of each module function.

File SaveLoadValues.lua:

```Lua
local M, module = {}, ...
function M.loadValue(fileName)
    package.loaded[module]=nil
    "your code"
end
function M.saveValue(fileName,commands)
    package.loaded[module]=nil
    "your code"
    end
end
```
To call the function use require("SaveLoadValues").loadValue("myfilename") or require("SaveLoadValues").saveValue("myfilename")

> Create your custom firmware at https://nodemcu-build.com/ I'm using only the following modules and with integer only: 

- cjson
- file
- gpio
- net
- node
- tmr
- wifi

> Always use local variables, i.e, local int=10

> Use UDP for LAN communication with others NodeMCU. Use TCP only to communicate with the user browser or WAN.

> Always compile your code and remove the .lua file (Right click -> compile at the lua file and after that Right click -> delete)

> Keep track of the free heap space with node.heap() command to check when some command is eating heap space.

NicolSpies, 2017-02-19:
Thanks, I will study it. 
I have been using the "flashmod" method by dpwhittaker. 

I suggest we take this offline as this forum is for development of NodeMCU. Would you have a problem of emailing me directly to continue the conversation.

marcelstoer, 2017-02-19:
No offence taken. Suggestions are good, PRs are even better 😉. 

An FAQ rewrite is long overdue, see #937. There's a lot of wisdom in it but you also have to understand [where it comes from](http://www.esp8266.com/wiki/doku.php?id=nodemcu-unofficial-faq) and put it's inception into a temporal context. Those were the times when ESP-01 modules were all we had.

> If you are trying to implement a user-interface or HTTP webserver in your ESP8266 then you are really abusing its intended purpose. 

I don't claim to know what the "intended purpose" is/was. The fact that lots of people are successfully running HTTP servers on ESP8266 are possibly an indication that we should dump that warning. 👍 

> Searching on google we found millions of pages talking about "No Memory" error

That's one of the problems of an information repository that never forgets, isn't it. Those pages are all from 2015 when NodeMCU was at 0.9.x I guess. Whenever I happen to come across such old claims I try to contact the author or leave a comment.

> I suggest we take this offline

I'm usually quick to close issues that "violate" #1010 but it'd be fine for me to continue here. If it results in improved FAQ pages then Yay! it was worth it.

gbechtinger, 2017-02-19:
> I have been using the "flashmod" method by dpwhittaker.

@NicolSpies flashmod It's different from volatile modules? Could you please point me to a page that explains how to do that? I can't find it.

NicolSpies, 2017-02-19:
It is complicated, [http://www.esp8266.com/viewtopic.php?f=19&t=1940&hilit=massive+memory&start=44](url), I suggest the following:

1. It is not directly related to the FAQ pages so lets take it offline.
2. Start reading the thread on page 12 ending on page one
3. I can mail you examples later

TerryE, 2017-02-19:
> the NodeMCU can be easily replace the Raspberry as a central server for IOT.

IMO, definitely no it can't.  Neither the ESP8266 non-OS SDK or the ESP32 RTOS are a fully featured OS with minimum acceptable security models.  I personally think it unwise to allow any direct access from these devices to the Internet represent a _material_ the security risk as last October's DDoS attack on the Dyn servers underlines.  The RPi runs a fully featured Linux OS, which can be locked down tightly.  It also has support a full LAMP stack, decent HDD and SSDs, etc.   

I am not saying that the RPi is the only server that can fulfil this role, nor am I saying that it can beat the ESP modules in their IoT sweet spot, but I feel it serious misguided to try to pretend that the ESP chips have capabilities that they simply do not.

marcelstoer, 2017-02-19:
> Neither the ESP8266 non-OS SDK or the ESP32 RTOS are a fully featured OS with minimum acceptable security models.

👍 yep, running a HTTP server omits [the first and most important line of defense](https://dzone.com/articles/have-we-forgotten-the-ancient-lessons).

> It is not directly related to the FAQ

@NicolSpies if it's an important or helpful concept (I've yet to read up on it) then it _should_ be part of our documentation.

closed 2017-02-19 by marcelstoer