nodemcu/nodemcu-firmware #1809: rtctime.get/set do not work proprelly when nested

thephantom1492, 2017-02-20:
### Expected behavior
Should be able to get the current time, add/substract some seconds, set back the new time

### Actual behavior
Very inconsistant results, no change or incorrect change.

### Test code
There is two examples here, the first I am unsure if it is the right way to do it or not, the second one is the main issue.

===== complete code start =====
wifi.setmode(wifi.STATION, false)
wifi.setphymode(wifi.PHYMODE_G)
wifi.sta.config("essid","pass") 

sntp.sync("192.168.10.1",
	function(sec, usec, server, info)
		rtctime.set(sec-18000, usec)   <=== this fail, unsure about the proper way, set the wrong time.
	end,
	true)

print(rtctime.get())

function gettime()
	tm = rtctime.epoch2cal(rtctime.get())
	print(string.format("%04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
end

gettime(); sec,usec=rtctime.get(); sec=sec-3600; rtctime.set(sec,usec);gettime()  <=== this make no change
===== complete code end =====

ex:
> gettime(); sec,usec=rtctime.get(); sec=sec-3600; rtctime.set(sec,usec);gettime() 
2017/02/19 23:56:38
2017/02/19 23:56:38

> gettime(); sec,usec=rtctime.get(); sec=sec-7200; rtctime.set(sec,usec);gettime() 
2017/02/20 00:09:14
2017/02/19 23:20:49

> gettime(); sec,usec=rtctime.get(); sec=sec-10800; rtctime.set(sec,usec);gettime() 
2017/02/19 23:21:47
2017/02/19 21:33:22

### NodeMCU version
https://nodemcu-build.com/ as of yesterday (master) and today (dev)

NodeMCU custom build by frightanic.com
        branch: dev
        commit: 0349c1e004cda7793e70463db121bde6b37d5e27
        SSL: false
        modules: adc,bit,cron,dht,encoder,file,gpio,http,i2c,net,node,ow,pwm,rtctime,sigma_delta,sntp,tmr,uart,websocket,wifi
 build  built on: 2017-02-20 03:40
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Hardware
Wemos d1 mini (that do not have the wemos marking)

ferdna, 2017-02-20:
===== complete code start =====
wifi.setmode(wifi.STATION, false)
wifi.setphymode(wifi.PHYMODE_G)
wifi.sta.config("essid","pass")

sntp.sync("192.168.10.1",
  function(sec, usec, server, info)
    print('sync', sec, usec, server)
  end,
  function()
   print('failed!')
  end
)

print(rtctime.get())

function gettime()
tm = rtctime.epoch2cal(rtctime.get())
print(string.format("%04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"]-5, tm["min"], tm["sec"]))
end

gettime();
===== complete code end =====

thephantom1492, 2017-02-20:
Ok ferdna, as I told you many many many many times on irc and you refuse to listen: you may claim to be a good programmer, but you refuse to admit that there may even be a bug. Let's break it down:
You also fail to acknowledge that you introduced an even bigger bug there (now the hour portion will go from -5 to 19 instead of 0-23).
You admitted to never have touched lua and hate it.
Also, you refuse to listen/acknowledge that the sntp is unrelated to the issue.
Also, since you never touched nodemcu, you don't know anything of how it work.

The bug I presented is about the rtctime.set that have an issue rolling back the time proprelly.
I even simplified the code, yet you refuse still to even agree that it have a bug or a documentation issue.

Simplified code:
function gettime()
    tm = rtctime.epoch2cal(rtctime.get())
    print(string.format("%04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
end

gettime(); sec,usec=rtctime.get(); sec=sec-3600; rtctime.set(sec,usec);gettime() 
2017/02/20 07:43:15
2017/02/20 07:43:15

Extra bonus: going forward seems to be fine:
gettime(); sec,usec=rtctime.get(); sec=sec+3600; rtctime.set(sec,usec);gettime() 
2017/02/20 07:45:14
2017/02/20 08:45:14

Another code example that somehow work:
rtctime.set(1487576564,0); gettime(); rtctime.set(1487576564-3600); gettime()
2017/02/20 08:54:18
2017/02/20 07:54:18

I'm suspecting a race condition between rtctime.get() and rtctime.set()

zelll, 2017-02-20:
Interesting.

`rtctime.set(1300000000, 500000);`
`rtctime.set(1500000000, 500000); rtctime.set(1400000000, 500000); print(rtctime.get());`
`rtctime.set(1500000000, 500000); rtctime.set(1600000000, 500000); print(rtctime.get());`

prints 
>1400004295	467608
>1600000000	500297

(1400004295.467608-1400000000.500000)-(1600000000.500297-1600000000.500000) = 4294.967311

2^32 us?

thephantom1492, 2017-02-20:
I don't understand your last math purpose, but a 32bits sound plausible, or not. My other tests seems to indicate that the issue is rolling back in time, but forward have no issue, so make the 32 bits theory seems improbable.

zelll, 2017-02-20:
> I don't understand your last math purpose
Forward works, so `rtctime.set(1600000000, 500000); print(rtctime.get());` takes 297 us. Then first print should print "1400000000 500297". But it prints "1400000000 500297"+2^32.

https://github.com/nodemcu/nodemcu-firmware/blob/master/app/include/rtc/rtctime_internal.h#L724
That's it. I think it's for sntp module which also calls this function.

thephantom1492, 2017-02-20:
It look like you found atleast part of the issue with that code. The time should and must be able to go backward as the clock can go too fast.

thephantom1492, 2017-02-21:
Small workaround for the issue I have...

sec,usec=rtctime.get()
sec=sec-3600
rtctime.set(0,0)
rtctime.set(sec,usec)

pjsg, 2017-02-22:
It turns out that the code really doesn't want time to go backwards. If you try and set the time backwards, then it really just runs the clock slow for a while until the time catches up. 

It isn't clear to me that this is always the right solution (or even that it gets it right for large differences). 

Some applications really don't want the time to go backwards -- maybe it makes sense to have a 'mode' bit that says whether this is allowable or not.

thephantom1492, 2017-02-22:
It do make sense to prevent a small 'few seconds' rollback, not big one. Probably that the proper way would be to add a force option AND a big jump exception to the no rollback. Like if more than a few minutes then allow the backward jump, but if only a few seconds then do fix it with the offset so it integrate it slowly.

pjsg, 2017-02-23:
If we can do it without having a force option, I would prefer it. My feeling is that less than (say) 10 seconds backwards can be handled with a slew. More than 10 seconds is a jump.

zelll, 2017-02-23:
@pjsg If rtc time is 1400000010 s+ 500000 us, which result we will get by calling `rtctime.set(1400000000, 499900)`? That's not good to guess.
I prefer `rtctime.set()` to be a force way, then add a `sntp.set()` to handle time in NTP way...

jmattsson, 2017-03-01:
Real time does not move backwards. Not in a way that's currently accessible to humanity at least. If you need support for jumping time backwards, which you really shouldn't, because the very point of the `rtctime` module is to enable accurate, forward-only time-keeping, there is a fairly easy workaround.

If you for whatever reason need to jump backwards, you could force the `rtctime` module to reinitialise by wiping it's "magic" rtcmem slot: `rtcmem.write32(0,0)` before your next call to `rtctime.set()` or `sntp.sync()` (which internally chains through to rtctime if compiled in).

djphoenix, 2017-03-01:
_Addition for @jmattsson words:_
I see you (@thephantom1492) trying to adjust synced time to constant seconds. I guess this is timezone adjustment?

thephantom1492, 2017-03-01:
yes timezone adjustement

djphoenix, 2017-03-01:
So I think the best way (with current API) to do it in Lua-level. Right now we haven't TZ functions in any module, but I think we _should_ have them... In future.

jmattsson, 2017-03-01:
I think @pjsg had some timezone stuff floating around? The accepted wisdom is that timezones are applied just before presentation, everything beforehand uses UTC to preserve sanity :)

thephantom1492, 2017-03-01:
I still beleive that the easiest place for the timezone would be with the sntp, allowing the clock to go backward there (maybe via a flag?) could be done easilly.

jmattsson, 2017-03-01:
Nonononono, (S)NTP only ever deals with UTC.

If timezone stuff goes anywhere near the `rtctime` module, it'll be the `rtctime.epoch2cal()` function.

marcelstoer, 2017-03-01:
Based on the comments so fare I vote to close this.

pjsg, 2017-03-09:
Yes -- I have some timezone stuff -- all written in lua. It processes the compiled timezone files that are commonly found on unix boxes. Is there a good place to put this kind of utility code?

jmattsson, 2017-03-09:
`lua_examples/`?

closed 2017-03-12 by marcelstoer