nodemcu/nodemcu-firmware #1810: Upgrade to next 2.x non-OS SDK

marcelstoer, 2017-02-21:
We were told that in about a month we'll get a new non-OS SDK with proper 16MB flash support, Yah!: https://github.com/espressif/esptool/issues/123#issuecomment-281256779

Upgrade to it once it's out. Even if Espressif sticks to the 1-month promise it'll be tight for [our next release](https://github.com/nodemcu/nodemcu-firmware/milestone/5). In that case I propose we shift the self-set April 1st deadline.

devsaurus, 2017-02-21:
That's great news indeed!

:+1: to shift the next drop (Sdk update will hopefully integrate seamlessly).

devsaurus, 2017-04-19:
We should have an eye on https://github.com/espressif/ESP8266_NONOS_SDK, I suppose.

marcelstoer, 2017-04-19:
Angus said this ten days ago at espressif/esptool#123:

> The ESP8266 non-OS SDK with support for 8MB & 16MB flash sizes is being tested now, ahead of a release. esptool.py support for these sizes should be available in the master branch shortly.

I've since been watching http://espressif.com/en/support/download/sdks-demos?keys=&field_type_tid%5B%5D=14 every other day or so. I didn't know Espressif moved the non-OS SDK to GitHub.
Thanks @projectgus and @igrr, nice move.

marcelstoer, 2017-05-05:
Yah! It finally landed: https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.1.0
And indeed there it says "Support 8MB & 16MB spi flash".  The list of changes is quite long but unfortunately they're missing references to the issue origin (Espressif BBS et.al.).
Stuff I find interesting:
- "Support long periods of deep sleep" - we need to update the `node.dsleep()` documentation!?
- "Fix http packet error" - unlikely to be related to the issues with our `http` module (#1721, #1707)?
- "lw IP: Add set and get dns server APIs" - any impact on us @djphoenix ?
- long list of WiFi fixes - @dnc40085 are you able to relate some of those to WiFi issues in our tracker? I didn't find anything that looked familiar (i.e. not obvious enough so that even I would spot it).

dnc40085, 2017-05-06:
>long list of WiFi fixes - @dnc40085 are you able to relate some of those to WiFi issues in our tracker?

None that I've been able to find.

devsaurus, 2017-05-06:
I would integrate the new SDK release within the next days if nobody else beats me to this. Plus some basic smoke tests. Aligning our 8MB & 16MB flash support can follow in a separate PR IMO.

Shall we continue to wget the release zip or migrate to a git submodule now that repo and tags are on github? I find the submodule setup of `dev-esp32` quite elegant after working with it for some time 😉

drawkula, 2017-05-06:
[X] — Git submodule.
;-)

marcelstoer, 2017-05-06:
Thanks Arnim 🥇 

> Aligning our 8MB & 16MB flash support can follow in a separate PR IMO.

👍 (but before the master drop)

> Shall we continue to wget the release zip or migrate to a git submodule

I'm all in favour of submodules but I'd do that _after_ the master drop. I want to get that out the door as soon as possible.

devsaurus, 2017-05-06:
> *after* the master drop. I want to get that out the door as soon as possible.

👍

devsaurus, 2017-05-07:
I got the firmware compiled with SDK 2.1.0 but look at continuous boot loops now:

> Fatal exception 0(IllegalInstructionCause):  epc1=0x402408f7, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x%

and then the output craps out until bootloader comes back with "ets Jan 8 2013".

epc1 points into `Cache_Read_Disable_2` and I haven't found any trace where to start debugging from.

@jmattsson any chance you can back up here?

jmattsson, 2017-05-07:
I don't think I've seen a `Cache_Read_Disable_2` in the past, I think that's a new function. My guess here would be that it's something that needs to be loaded into IRAM, and we're not. That would definitely yield the IllegalInstruction exception as the core tries to read instructions from an unmapped flash. Check the linker script/map file?

devsaurus, 2017-05-08:
Thanks Johny, this pointed to the right spot. `Cache_Read_Disable_2` was mapped to irom0 even though libmain.a is supposed to end up in iram0.

The output of `nm -a sdk/esp_iot_sdk_v2.1.0/lib/libmain.a`
> 00000000 t .irom0.literal
00000000 t .irom0.text
00000000 t .irom.text
00000000 t .literal.Cache_Read_Disable_2
00000000 t .literal.Cache_Read_Enable_2
00000000 t .literal.flash_gd25q32c_enable_QIO_mode
00000000 t .literal.flash_gd25q32c_read_status
00000000 t .literal.flash_gd25q32c_write_status
00000000 t .literal.spi_flash_check_wr_protect
00000000 t .literal.spi_flash_enable_qmode
00000000 t .literal.spi_flash_erase_sector
00000000 t .literal.spi_flash_get_id
00000000 t .literal.spi_flash_get_unique_id
00000000 t .literal.spi_flash_issi_enable_QIO_mode
00000000 t .literal.spi_flash_read
00000000 t .literal.spi_flash_read_status
00000000 t .literal.spi_flash_write
00000000 t .literal.spi_flash_write_status
00000000 t .literal.user_spi_flash_dio_to_qio_pre_init

lead me to the assumption that our current statements in nodemcu.ld don't catch each and every aspect. My remedy was to add some wildcards in https://github.com/devsaurus/nodemcu-firmware/commit/73d3b6654c1bec433e2167f7182c6201e820241b.
Now I get the Lua prompt.

Is this the correct solution?
Should I add such wildcards elsewhere, too?

jmattsson, 2017-05-08:
Ah, they've started compiling (some?) stuff with `-ffunction-sections`. I'd suggest
```
*libmain.a:*(.literal .literal.* .text .text.*)
```
as the more canonical approach to it, though yours would likely work just as well :)

devsaurus, 2017-05-08:
Oh, nice. This gives us a finer granularity for deciding which functions/sections are placed in iram? And force the rest of the library content to irom, optimizing RAM usage. Theoretically at least, since we don't have clear information in which context functions are called...

jmattsson, 2017-05-08:
There might be more hints in the default SDK linker file for 2.1 then

djphoenix, 2017-05-08:
I'll take a look on linking script this evening/night.

devsaurus, 2017-05-08:
Ok, I'll make a PR to collect specific feedback end have scripts modified eventually.

devsaurus, 2017-05-08:
Transparent support of >4MB boards is in the pipe at https://github.com/devsaurus/nodemcu-firmware/tree/align_8mb_16mb.
Wemos D1 mini pro works on standard flash upgrade process (flash erase, image download, auto spiffs format) using latest esptool.py ✌️

PR #1962 (2017-05-11): Upgrade to SDK 2.1.0.
Fixes #1810.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This updates the SDK to [upstream 2.1.0 release](https://github.com/espressif/ESP8266_NONOS_SDK/releases/tag/v2.1.0).
Notable changes:
- Some ROM functions are now defined in sdk headers and had to be [... cut]
closed 2017-05-15 by marcelstoer