nodemcu/nodemcu-firmware #1811: MQTT keep-alive packets corrupted?

joysfera, 2017-02-21:
Has someone else noticed that MQTT connection breaks after some time?

### Expected behavior
MQTT connection is kept alive pretty much forever

### Actual behavior
After certain time (hours or few days of a well working MQTT connection) the keep-alive MQTT packets suddenly upset Mosquitto broker so it disconnects the MQTT NodeMCU Lua client immediately:
```Socket error on client abcdef, disconnecting.```

I have implemented my own reconnecting code (since the stock NodeMCU one is too dangerous so I don't use it) but even after my MQTT client reconnects and the keepalive time passes the first keep-alive  packet is again reason why Mosquitto kicks out my NodeMCU client immediately. 

### NodeMCU version
Feb 2 stable

NicolSpies, 2017-02-21:
I experienced the MQTT connection to becoming "unresponsive" way back #1406. I never got the Socket error but it sound similar to what I experienced. I tested with CloudMQTT also using Mosquitto.

I devised some workarounds and never spend time to determine if the problem was fixed in later NodeMCU releases.

joysfera, 2017-02-21:
I'm happy I tracked it down to the empty ping packets. Should be easy to fix now?

NicolSpies, 2017-02-21:
I hope so, it has been bugging me for a long time but I am dependent on the experts to fix it.

joysfera, 2017-02-21:
We're talking about this code in modules/mqtt.c:
```
          uint8_t temp_buffer[MQTT_BUF_SIZE];
          mqtt_msg_init(&mud->mqtt_state.mqtt_connection, temp_buffer, MQTT_BUF_SIZE);
          NODE_DBG("\r\nMQTT: Send keepalive packet\r\n");
          mqtt_message_t* temp_msg = mqtt_msg_pingreq(&mud->mqtt_state.mqtt_connection);
          msg_queue_t *node = msg_enqueue( &(mud->mqtt_state.pending_msg_q), temp_msg,
                              0, MQTT_MSG_TYPE_PINGREQ, (int)mqtt_get_qos(temp_msg->data) );
          mud->keepalive_sent = 1;
          mud->keep_alive_tick = 0;     // Need to reset to zero in case flow control stopped.
          mqtt_send_if_possible(mud->pesp_conn);
```
I have yet to understand the two functions called there to see if the msg.length is set correctly or if it can end up zero.

nwf, 2017-02-22:
If there's packet corruption inside the MQTT library, that could also explain #1773 which remains mysterious to me.

NicolSpies, 2017-02-22:
@marcelstoer, I really believe this might be the same problem I reported last August #1406.
If I recall my workaround was to do a "dummy" publish on a regular basis to keep the client responsive. This would have eliminated the dependence on the keep-alive packets if I am correct ?

marcelstoer, 2017-02-22:
Didn't we establish back then that @djphoenix 's PR fix this?

NicolSpies, 2017-02-22:
No, @djphoenix concurred that his patch did not fix the issue but because the root cause couldn't be established the exercise went stale. This definitely looks like the smoking gun.

marcelstoer, 2017-02-23:
@pjsg any idea?

georeb, 2017-06-18:
I am also getting keep-alive packet issues.

I am trying to connect to a MS Azure IoT Hub and the connection is accepted, there is no problem and I receive cloud-device messages.

However, when either a keep-alive packet is due to be sent or i try to publish, the broker disconnects me! I'm using ESP-12F on latest NodeMCU

This is the debug information, after a keep-alive packet is sent...

```
enter mqtt_socket_publish.
Sent: 47
send_if_poss, queue size: 1
publish, queue size: 1
leave mqtt_socket_publish.
> enter mqtt_socket_sent.
sent1, queue size: 1
Data published successfully.
send_if_poss, queue size: 0
sent2, queue size: 0
leave mqtt_socket_sent.
client's data invalid protocol
Reason:[-0x7880]
enter mqtt_socket_reconnected.
Error Connecting: -5
enter mqtt_socket_disconnected.
MQTT Disconnected.
leave mqtt_socket_disconnected.
leave mqtt_socket_reconnected.

 DYNARR_DBG(dynarr_remove): element(3fff0a88) removed from array
```

This is the debug information, after a publish is exectuted...

```
MQTT: Send keepalive packet
Sent: 2
send_if_poss, queue size: 1
keep_alive_tick: 0
leave mqtt_socket_timer.
enter mqtt_socket_sent.
sent1, queue size: 1
send_if_poss, queue size: 0
sent2, queue size: 0
leave mqtt_socket_sent.
client's data invalid protocol
Reason:[-0x7880]
enter mqtt_socket_reconnected.
Error Connecting: -5
enter mqtt_socket_disconnected.
MQTT Disconnected.
leave mqtt_socket_disconnected.
leave mqtt_socket_reconnected.

 DYNARR_DBG(dynarr_remove): element(3fff0a88) removed from array
```

Please can you help @marcelstoer ? Thankyou

marcelstoer, 2017-06-19:
@georeb I know a thing or two about the NodeMCU ecosystem but I'm not a _firmware_ developer. I'm quite convinced your issue is unrelated to this one. Maybe one of the MQTT experts (@joysfera, @NicolSpies and primarily @pjsg) can confirm this.
`client's data invalid protocol` comes from https://github.com/nodemcu/nodemcu-firmware/blob/master/app/mbedtls/app/espconn_mbedtls.c#L434 and sounds more related to the notorious #1707. I wish @djphoenix had more time to hunt that one down.

joysfera, 2017-06-19:
FYI, I started sending my own keep-alive packets (I simply send a MQTT message containing node.heap() every minute) and didn't have any MQTT issues since then.
I even got MQTT working with TLS/SSL once but had to give up on that due to lack of memory.

NicolSpies, 2017-06-19:
Yes, exactly what my workaround since last year as commented earlier in this thread.

TTasuja, 2018-06-05:
@georeb Did you get publishing working for MS Azure IoT Hub? I am experiencing exactly the same issue. MQQTS connection leaves me 16K of heap (I have no other LUA code running). I wonder if this might be the issue of publishing failure. I have publishing working only 10/90% of time. Need to dig even deeper in mbedtls stack..

closed 2019-08-04 by stale[bot]