nodemcu/nodemcu-firmware #1812: net.multicastJoin causes "illegal value" error in net.socket:send

FrankX0, 2017-02-21:
Using net.socket:send in combination with multicastJoin causes an error on the latest builds.
This used to work correctly, but seems broken after latest master drop.
### Expected behavior
No error
### Actual behavior
```Lua
PANIC: unprotected error in call to Lua API (udp.lua:4: illegal value)
```
### Test code
Use below code and send anything to the listen port and address.
```Lua
net.multicastJoin(wifi.sta.getip(), '239.255.255.250')

local function response(connection)
    connection:send(2000,'239.255.255.250',"Hello world")
end

UPnPd=net.createServer(net.UDP)
--UPnPd = net.createUDPSocket()
UPnPd:on("receive", response)
UPnPd:listen(2000,'239.255.255.250')
```
This results in:
```Lua
PANIC: unprotected error in call to Lua API (udp.lua:4: illegal value)
```

### NodeMCU version
Latest build (based on SDK 2.0.0)
Build from last December (based on SDK 1.5.4.1)
### Hardware
Any (NodeMCU v2/v3, Wemos D1)

marcelstoer, 2017-02-22:
As per the latest release notes there were some changes to UDP support. However, since you already have `--UPnPd = net.createUDPSocket()` commented in your code I assume you're aware that `net.createServer(net.UDP)` is now deprecated.

Are you saying that [our UDP example](http://nodemcu.readthedocs.io/en/latest/en/modules/net/#netudpsocketsend) doesn't work in combination with `multicastJoin`?

FrankX0, 2017-02-23:
Correct.
The test code is a stripped down version of the actual code (UPnP).
That was running fine on the older firmware (with the deprecated functions).
Through experimenting I found out that using `multicastJoin `triggered the error.

djphoenix, 2017-02-23:
@FrankX0 thanks for report. I haven't tested most of Multicast features, so I'll play with them at this weekend.
But in the first, is it works for *send* messages to multicast address? And so, are them received if socket bound to 0.0.0.0?

FrankX0, 2017-02-23:
There is another, even simpler way, to reproduce:
```Lua
UPnPd = net.createUDPSocket()
UPnPd:listen(2000,'239.255.255.250')
UPnPd:send(2000,'239.255.255.250',"Hello world")
```
This fails in the same way (illegal value).
It fails due to the combination of `listen()` and `send()` and multicast messages.
Removing `listen()` removes the error.

djphoenix, 2017-03-16:
@FrankX0 played for some time. So `listen` bounds socket to some address, what used for message that being sent. So there are no **interface** with multicast IP, and _this_ produces "invalid address" error.

In my case, `listening` on `0.0.0.0` with `sending` to mutlicast address works properly. Objections from you?

FrankX0, 2017-03-17:
Thanks @djphoenix!
Changing the listen address from the broadcast address to `0.0.0.0` not only stopped the crashing, now also my application (UPnP) is working again!

I think we should document this somewhere. Anyone suggestions?

gepeng1983, 2017-03-18:
@djphoenix I followed this thread and got the listening working. Now the multicast send seems not reaching my other computers in the same subnet, as captured by wireshark. The code is a modified version of https://github.com/pastukhov/nodemcu-ssdp as in the pastukhov/nodemcu-ssdp#1 problem.

FrankX0, 2017-03-22:
This is the modified SSDP application I am using:
```lua
net.multicastJoin(wifi.sta.getip(), "239.255.255.250")

local ssdp_notify = "NOTIFY * HTTP/1.1\r\n"..
"HOST: 239.255.255.250:1900\r\n"..
"CACHE-CONTROL: max-age=100\r\n"..
"NT: upnp:rootdevice\r\n"..
"USN: uuid:c5baf4a1-0c8e-44da-9714-ef01234"..string.format("%x",node.chipid()).."::upnp:rootdevice\r\n"..
"NTS: ssdp:alive\r\n"..
"SERVER: NodeMCU/20150415 UPnP/1.1 ovoi/0.1\r\n"..
"Location: http://"..wifi.sta.getip().."/DimmableLight.xml\r\n\r\n"

local ssdp_response = "HTTP/1.1 200 OK\r\n"..
"Cache-Control: max-age=100\r\n"..
"EXT:\r\n"..
"SERVER: NodeMCU/20150415 UPnP/1.1 ovoi/0.1\r\n"..
"ST: upnp:rootdevice\r\n"..
"USN: uuid:c5baf4a1-0c8e-44da-9714-ef01234"..string.format("%x",node.chipid()).."\r\n"..
"Location: http://"..wifi.sta.getip().."/DimmableLight.xml\r\n\r\n"

notifyCount = 0

local function response(connection, payLoad, port, ip)
        if string.match(payLoad,"M-SEARCH") then
            connection:send(port,ip,ssdp_response)
        end
end

tmr.alarm(3, 10000, 1, function()
    notifyCount = notifyCount + 1
    if notifyCount == 3 then
        tmr.stop(3)
        notifyCount = nil
        ssdp_notify = nil
    else
        UPnP = net.createUDPSocket()
        UPnP:send(1900,'239.255.255.250',ssdp_notify)
        UPnP:close()
        UPnP = nil
        notify = nil
    end
end)

UPnPd = net.createUDPSocket()
UPnPd:on("receive", response )
UPnPd:listen(1900,"0.0.0.0")

```

djphoenix, 2017-03-23:
@FrankX0 why'd you recreate UPnP UDP socket for each outgoing frame? You already have one (UPnPd), you may use it to send frames too.

gepeng1983, 2017-03-23:
@FrankX0 @djphoenix Neither way gets the broadcast UDP packets sent, as seen in wireshark. (While other UPNP devices in the same router successfully sent UDP packets for NOTIFY.)

FrankX0, 2017-03-23:
@gepeng1983 I do see the broadcasts in Wireshark, as I do of e.g. my router and printer.

@djphoenix you are correct. I just used this code without further changes as it was working.
I also noticed that the NOTIFY packets should be sent regularly. So resulting code:

```lua
net.multicastJoin(wifi.sta.getip(), "239.255.255.250")

local ssdp_notify = "NOTIFY * HTTP/1.1\r\n"..
"HOST: 239.255.255.250:1900\r\n"..
"CACHE-CONTROL: max-age=100\r\n"..
"NT: upnp:rootdevice\r\n"..
"USN: uuid:c5baf4a1-0c8e-44da-9714-ef01234"..string.format("%x",node.chipid()).."::upnp:rootdevice\r\n"..
"NTS: ssdp:alive\r\n"..
"SERVER: NodeMCU/20150415 UPnP/1.1 ovoi/0.1\r\n"..
"Location: http://"..wifi.sta.getip().."/DimmableLight.xml\r\n\r\n"

local ssdp_response = "HTTP/1.1 200 OK\r\n"..
"Cache-Control: max-age=100\r\n"..
"EXT:\r\n"..
"SERVER: NodeMCU/20150415 UPnP/1.1 ovoi/0.1\r\n"..
"ST: upnp:rootdevice\r\n"..
"USN: uuid:c5baf4a1-0c8e-44da-9714-ef01234"..string.format("%x",node.chipid()).."\r\n"..
"Location: http://"..wifi.sta.getip().."/DimmableLight.xml\r\n\r\n"

local function response(connection, payLoad, port, ip)
    if string.match(payLoad,"M-SEARCH") then
        connection:send(port,ip,ssdp_response)
    end
end

tmr.alarm(3, 10000, 1, function()
    UPnPd:send(1900,'239.255.255.250',ssdp_notify)
end)

UPnPd = net.createUDPSocket()
UPnPd:on("receive", response )
UPnPd:listen(1900,"0.0.0.0")
```

gepeng1983, 2017-03-24:
@FrankX0 What version of nodemcu are you using? Confirmed your code doesn't work on my nodemcu compiled on nodemcu-build.com a few days earlier.

gepeng1983, 2017-03-24:
@djphoenix 

Found the problem.

A simple script with UPnP code works without my main code. That allowed me to question part of the main code. It turned out that when I do enduser_setup.manual, in order to set the AP to a name, I set the wifi in wifi.STATIONAP; later, multicast UDP doesn't send in STATIONAP mode.

Changing back STATIONAP to STATION somewhere in the setup code allowed the UDP multicast sent.

Please close case

closed 2017-03-24 by FrankX0