nodemcu/nodemcu-firmware #1815: ws2812.init() cause esp12-e reboot

sivix, 2017-02-23:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
ws2812.init()  should enable GPIO2 for communication
### Actual behavior
=ws2812.init()

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26896, room 16 
tail 0
chksum 0x5e
load 0x3ffe8000, len 2424, room 8 
tail 0
chksum 0x02
load 0x3ffe8978, len 136, room 8 
tail 0
chksum 0xf1
csum 0xf1
����n��r��n|�llll`��r�l�l�

NodeMCU custom build by frightanic.com
	branch: master
	commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
	SSL: true
	modules: adc,enduser_setup,file,gpio,mqtt,net,node,pwm,tmr,uart,wifi,ws2812,tls
 build 	built on: 2017-02-19 17:27
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
ws2812.init()
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
NodeMCU custom build by frightanic.com
	branch: master
	commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
	SSL: true
	modules: adc,enduser_setup,file,gpio,mqtt,net,node,pwm,tmr,uart,wifi,ws2812,tls
 build 	built on: 2017-02-19 17:27
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.
simple chip esp8266-12e, GPIO15 to GND, CH_PD to VCC

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

devsaurus, 2017-02-23:
Same instant crash here on a build from current `dev` 😞 
Thanks for reporting @sivix!

devsaurus, 2017-02-23:
That's a simple one: `ws2812_init()` doesn't return number of items pushed onto the stack. Wonder why this tunnels the prototype checks. Nevertheless, PR will follow soon.

@sivix just call `ws2812.init()` without enclosing print. The function isn't supposed to return status anyhow.

closed 2017-02-24 by marcelstoer

pjsg, 2017-02-24:
Is there another compiler warning that we should enable to catch this?

Closed #1815.

devsaurus, 2017-02-27:
> Is there another compiler warning that we should enable to catch this?

None that I'm aware of. I'd see the possibility to add a dedicated function pointer to the union in [lobject.h](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/lua/lobject.h#L83) and have the [LFUNCVAL macro](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/lua/lrotable.h#L13) assign the function to that pointer. Should enable the compile to perform checks during compile time. Don't know if it's worth the effort, though.

PR #1816 (2017-02-24): Fix missing return code for ws2812_init()
Fixes #1815.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Like all Lua functions, `ws2812_init()` must return number of pushed elements onto stack.

Small corrections for `ws2812.init()` API doc formatting.