nodemcu/nodemcu-firmware #1817: Bug in node.compile()

EgorOA, 2017-02-23:
node.compile() makes silent system restart while compiling chunk with the number of **0x7fffffff** (or decimal equivalent).

so have to do something like this in code to work with such number

> local mask = 0x7ffffffe
> mask = mask + 1

---------------------------------------------------------
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 0349c1e004cda7793e70463db121bde6b37d5e27
	SSL: false
	modules: adc,bit,bme280,dht,enduser_setup,file,gpio,i2c,mqtt,net,node,ow,rtctime,sntp,spi,tmr,uart,ucg,wifi
 build 	built on: 2017-02-17 20:39
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

FrankX0, 2017-02-24:
Duplicate of #1691

devsaurus, 2017-02-24:
Thanks @FrankX0!

closed 2017-02-24 by devsaurus