nodemcu/nodemcu-firmware #1818: ws2812.init(); the BLUE led onboard of the ESP8266-12e is On

sivix, 2017-02-24:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
ws2812.init() just initialize the communication with the ws2812.

### Actual behavior
the Blue led onboard of the ESP8266-12E is fixed LIGHT ON... altering the coloring emitted by any circuits... there is a way to turn the BLUE-LED onboard Off?

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
NodeMCU custom build by frightanic.com
	branch: master
	commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
	SSL: true
	modules: adc,enduser_setup,file,gpio,mqtt,net,node,pwm,tmr,uart,wifi,ws2812,tls
 build 	built on: 2017-02-23 12:48
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
Starting SmartButton

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

ESP8266-12e connected to ws2812 using GPIO2.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

Destroy the onboard led?

8<------------------------ END FEATURE REQUEST --------------------------------

FrankX0, 2017-02-24:
The LED on the ESP-12E is hardwired to GPIO2 (refer to the [schematics](http://www.esp8266.com/wiki/lib/exe/fetch.php?media=schematic_esp-12e.png)).
So any activity (low level) on GPIO2 will turn the LED on.
So disabling is only possible by removing the resistor and/or LED. Sorry.

devsaurus, 2017-02-24:
Also see #1799.

closed 2017-02-24 by marcelstoer