nodemcu/nodemcu-firmware #1822: EUS. onConnected never called

a3-kovalyov, 2017-02-27:
### Expected behavior
After set wifi credentials `onConnected` should be called.

### Actual behavior
`onConnected` never called. After manual reset device successfully connects to wifi

### Test code
```Lua
wifi.setmode(wifi.STATIONAP)
wifi.ap.config({ssid="MyPersonalSSID", auth=wifi.OPEN})
enduser_setup.manual(true)
enduser_setup.start(
  function()
    print("Connected to wifi as:" .. wifi.sta.getip())
  end,
  function(err, str)
    print("enduser_setup: Err #" .. err .. ": " .. str)
  end
);
```
"Connected to wifi as: X.X.X.X" never printed

### NodeMCU version
NodeMCU custom build by frightanic.com
branch: master
commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
SSL: false
modules: apa102,bme280,bmp085,cjson,cron,crypto,dht,encoder,enduser_setup,file,gpio,http,mdns,mqtt,net,node,spi,tmr,uart,websocket,wifi,wps
build 	built on: 2017-02-09 18:49
powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Hardware
ESP8266 Amica https://ru.aliexpress.com/item/New-Wireless-module-NodeMcu-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266-with-pcb-Antenna/32266017059.html

FrankX0, 2017-03-17:
Sorry, cannot reproduce your findings.

young310, 2017-03-23:
@a3-kovalyov 
Please give it a try to remove the first two lines, and you should obtain the ip.

rsabin, 2017-03-28:
It seems that setting `enduser_setup.manual(true)` causes the callbacks routines never to be called.
I have this issue too.

Howrever thats doesn't matter because `wifi.ap.config({ssid="MyPersonalSSID", auth=wifi.OPEN})` doesn't work neither way. Your ssid will be something like `SetupGadget_XXXXXX` after 1-2 resets. 
If you want to change the network name, you need to clone the project, edit `user_config.h` and recompile yourself. Then just use `enduser_setup.manual(false)` and move on.

jmattsson, 2017-05-04:
Tagging @jfollas for possible insight.

jfollas, 2017-05-04:
When I was last in the code, I did notice that the manual mode was not invoking the success callback. I think I worked on a fix for that, but then shelved the changes that I was working on due to a bug in the SDK that Espressif hasn't addressed yet (Channel Switch Announcements issue that seemed to be interfering with the ability for your machine to find the SoftAP after saving wi-fi config if a radio channel change is required).

http://bbs.espressif.com/viewtopic.php?f=66&t=3328

As an aside - why do so many people want to use manual mode instead of the automatic mode? Is it just so you can set the SSID from Lua, or is there another reason?  Maybe there's an opportunity to expose that functionality to Lua.

closed 2019-06-21 by stale[bot]