nodemcu/nodemcu-firmware #1823: [ucglib] Include support for ILI9225?

ne0h, 2017-02-28:
### Missing feature
Support for SPI TFT ILI9225. I guess the usage is similar to already supported TFTs like ILI9341 or ILI9163.

### Justification
I would like to use my TFT. I have already used it with Arduino Mega and this library: https://github.com/Nkawu/TFT_22_ILI9225. Hope this helps.

Thanks in advance!
Max

devsaurus, 2017-02-28:
We're just integrating the upstream [Ucglib project](https://github.com/olikraus/ucglib) and there's no development of drivers or else here. So please check with this project about plans to add ILI9225.

Updating the NodeMCU firmware will be quick once the display is added in Ucglib.

tomtor, 2017-02-28:
I did a simple LUA only port:

https://github.com/tomtor/ESP-8266/blob/master/tft.lua

marcelstoer, 2017-02-28:
Thanks Tom. Someone looking for ILI9225 and NodeMCU will certainly find this helpful.

closed 2017-02-28 by marcelstoer

ne0h, 2017-03-01:
Thanks a lot for the information!
I am going to inform the ucglib project directly.