nodemcu/nodemcu-firmware #1825: enduser should allow extra params to be passed and store them for retrieval

rvowles, 2017-03-01:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

### Caveat: I have already built and tested this functionality with mods to the enduser module. It needs conforming to the standards of the rest of the project, but I am logging this to determine if what I see is a real need is just my imagination. 

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
Ideally when setting up your esp, you want to be able to configure it all in one go (or at least everything that makes sense) rather than resorting to hardware switches or similar - especially if you have a number of similar ones. I propose allowing people, in their replacement enduser pages to be able to pass more parameters to the esp and be saved for connection success and re-awakening.

e.g.
url -v "http://example.com/setwifi?wifi_ssid=Halls%20Of%20The%20Light%20Queen&wifi_password=MyPassword&location=master&fish=pox%20pox%20pox&silly=put%3Ety"

should connect to the wifi and create a file (say enduser.json) that holds:

{"location":"master","fish":"pox pox pox","silly":"put>ty"}

### Justification
It makes the esp much easier to setup for the general use case and this module far more useful.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

Sure, but they are all very painful

8<------------------------ END FEATURE REQUEST --------------------------------

joysfera, 2017-03-01:
FYI, I don't use the end-user setup because it doesn't allow me to specify a hostname. I believe that user needs to specify hostname of the node before letting it connect to the AP for the first time.
Maybe that this feature request could allow me to add a "Hostname:" entry to the dialog thus I could start using it.
My current solution is basically reimplementation of this functionality in plain Lua which is memory hungry, not pretty and worst of all - doesn't provide fake DNS replies thus user must type in the exact IP address of the node.

rvowles, 2017-03-01:
It would let you do exactly that.

marcelstoer, 2017-03-01:
That's a dup of #1302.

closed 2017-03-01 by marcelstoer

rvowles, 2017-03-01:
Thanks @marcelstoer - I searched for enduser but not wish.

rvowles, 2017-03-01:
I mean eus. Auto correct.