nodemcu/nodemcu-firmware #1826: adc.readvdd33() on wemos d1 mini pro always gives 65535

Tuxdiver, 2017-03-01:
### Expected behavior
Calling `adc.readvdd33()` should return the voltage after setting `adc.force_init_mode(adc.INIT_VDD33)` and possibly restarting

### Actual behavior
Call of `adc.readvdd33()` always returns 65535, `adc.read(0)` returns analog value around 7.

### Test code
```Lua
print(adc.force_init_mode(adc.INIT_VDD33))
print(adc.readvdd33())
print(adc.read(0))
```

gives:
```
false
65535
7
```

### NodeMCU version
Current own build by docker image:
NodeMCU 2.0.0 build unspecified powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Hardware
Wemos d1 mini Pro

### Possible reason
Could the changed memory layout for modules with more than 4MB flash cause the problem? Maybe the writing to memory position 107 is at the wrong position in the flash?
adc.c:
`  uint32 init_sector = flash_safe_get_sec_num () - 4;`
It does write something, because calling `adc.force_init_mode` with changing mode returns true in the first call and false after that. But maybe it is not in the right block.

devsaurus, 2017-03-01:
Your analysis is most likely correct, thanks for reporting!
Seems I missed this instance - will have a look.

devsaurus, 2017-03-01:
@Tuxdiver I was able to reproduce this issue. Fix is contained in the linked PR.

jmattsson, 2017-03-02:
Fix merged into dev.

PR #1827 (2017-03-02): Use flash size byte to determine the location of the init data for byte 107.
Fixes #1826.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

The current method to determine the location of the init data sector for byte 107 assumes that it's at offset -4 from the end of the flash. This logic breaks in case the current workaround for >4MB modules is present wh [... cut]
closed 2017-03-02 by jmattsson