nodemcu/nodemcu-firmware #1828: memory leak in net.createServer

stephanMettler, 2017-03-03:
Leaks in current dev and master

**Older Branches do not leak!
for example this old build from custom-firmware-builder does not leak:
branch: master, commit: 81ec3665cb5fe68eb8596612485cc206b65659c9**

### Test code
```Lua
srv=net.createServer(net.TCP)
srv:listen(80,function(conn)
	conn:on("receive", function(client,request)
		client:send('hello')
	end)
	conn:on("sent", function (c) c:close() end)
end)
```

devsaurus, 2017-03-03:
Does heap recover after some time like shown in https://github.com/nodemcu/nodemcu-firmware/pull/1379#issuecomment-268074953?

marcelstoer, 2017-03-03:
If it turns out to be another lwIP `TIME_WAIT` issue then we should document that and/or lower the value.

FrankX0, 2017-03-03:
I would opt for (drastically) lowering the timeout before releasing the memory.
Several people already came across this issue, including myself.
I have a web-server application with a self-refreshing page, refreshing every 10 seconds. This now causes the firmware to crash. Also performing some page refreshes by te user can now easily crash the server.
I am wondering why there is such a long timeout before releasing the memory, and what would be a better value, or even instant?

stephanMettler, 2017-03-03:
@devsaurus I am not sure if the heap recovers after > 2Minutes.
But the leak renders the whole net.createServer function useless (for my part)
Thx for referencing the other comments!!

marcelstoer, 2017-03-04:
Worth noting that other SDKs had their issues with this as well: https://github.com/SuperHouse/esp-open-rtos/issues/101

djphoenix, 2017-03-06:
Superseeded by #1836

closed 2017-03-06 by djphoenix