nodemcu/nodemcu-firmware #1835: "address in use" with net.createServer

joysfera, 2017-03-06:
### Expected behavior
no PANIC

### Actual behavior
PANIC: unprotected error in call to Lua API (wificonfig.lua:43: address in use)

### Test code
```Lua
39: wifi.setmode(wifi.STATIONAP)
40: wifi.sta.getap(copyAPs)
41: wifi.ap.config(cfg)
42: srv=net.createServer(net.TCP)
43: srv:listen(80,function(conn)
```

### Some background
My application runs web server in WiFi STATION mode normally but when users push a button for wifi reconfiguration the normal web server is shut down, WiFi is reconfigured to STATIONAP mode, wifi.sta.getap() is run (to fetch the list of available WiFi APs) and then a new web server is started.

Not sure since when (but I suspect the Feb 2 master drop) I started getting the above mentioned error "address in use". This used to work OK in January, I'd swear.

Any idea how to work around this? I made sure the previous server was indeed shut down cleanly so there shouldn't be any address in use. Or is it related to ```wifi.sta.getap()```? I have no idea. Please advise! Thanks.

### NodeMCU version
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
	SSL: false
	modules: bit,cjson,crypto,file,gpio,http,mdns,mqtt,net,node,ow,rtcmem,rtctime,sntp,tmr,uart,wifi
 build 	built on: 2017-02-02 09:20
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
```

### Hardware
Wemos D1 Mini and also pure ESP8266-07

djphoenix, 2017-03-06:
This is for sockets in `TIME_WAIT` status, see #1828 / #1642.
As a workaround - you may not close server but just reject (instantly close) incoming sockets when they're unexpected.

joysfera, 2017-03-06:
@djphoenix I don't understand. The previous net.server was running on 192.168.1.100 (a client IP) while now I need to start a new net.server on 192.168.4.1 (the usual ESP8266 configuration IP address).
Are you saying that it's OK to keep the net.server running while reconfiguring the WiFi mode and changing the IP address? That hardly makes sense...

djphoenix, 2017-03-06:
Yep, this is OK because *server* instances are bound to `0.0.0.0` address by default, and may be kept alive when interfaces reconfiguring.

joysfera, 2017-03-06:
that's quite misleading error message then ("address in use"). I'll give it a try, though. Thank you.

Sounds like the TIME_WAIT should be shortened to less than 1 second immediately, even in the master (stable) branch.

djphoenix, 2017-03-06:
TIME_WAIT is mechanism of TCP to keep track for incoming frames that may be (re)sent from remote side **after** socket becomes closed. So there may be reason to keep them in-memory for some long time, but of course we need to avoid this breaking logic.
I created issue #1836 to track it because now this notes are spread at many places...

closed 2017-03-06 by djphoenix

joysfera, 2017-03-06:
You closed it real quick. Please note @djphoenix that if I try to reuse the existing net.server I get the following error: ```wificonfig.lua:43: already listening```
Apparently I cannot set a new listen hadler but I cannot reuse the previous one because it's in a completely different file. Is there a way of removing old listen handler? The documentation doesn't suggest anything...

djphoenix, 2017-03-06:
Unfortunately you can't *replace* listen handler, but you may include some logic in one (assigned in first `listen`) and switch between them as needed.

joysfera, 2017-03-06:
Actually I need to kill the first one to free some memory. Yes, I could split things in even more modules and load them appropriately using the dofile() but this is getting rather complicated.
Please shorten the TIME_WAIT instead. It would fix these issues instantly, if I understand it properly. The TIME_WAIT does not need to be higher than worst ping time (doubled), as I have read somewhere...

djphoenix, 2017-03-06:
I will work on fixes (maybe - partially) for #1836 in short time. I hope that one of three enumerated ways may be simple and effective - but not promise that.

aditgupta100, 2017-05-06:
I'm getting the same error, i'm trying to create a local http server and also listening to a server page on port 80. Here's my pseudo code :

wifi.setmode(wifi.STATION)
wifi.sta.config("SSID","password")
wifi.sta.connect()

tmr.alarm(1,10000, 1, function()
    if (wifi.sta.getip() == nil) then
        print("IP unavaiable, Waiting...")
    else
        foo()
		local_server()
    end
end)

function foo()
        conn = nil
        conn=net.createConnection(net.TCP,0)
        conn:on("receive", function(conn, payload)
            print("payload : "..#payload);
        end)
        conn:connect(80,"server.co.in")
        conn:on("connection", function(conn, payload)
            conn:send("GET /mypage?id=deadbeef HTTP/1.0\r\nHost: server.co.in\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n") end)

        conn:on("sent",function(conn)
                print("Closing server connection")
                conn:close()
            end)
end

function local_server()
    srv=net.createServer(net.TCP)
    srv:listen(80,function(conn)
        conn:on("receive", function(client,request)
			local buf = "Hello world";
            client:send('HTTP/1.0\n\n')
            client:send('<!DOCTYPE HTML>\n')
            client:send('<html>\n')
            client:send('<head><meta  content="text/html; charset=utf-8">\n')
            client:send('<title>Local server</title></head>\n')
            client:send('<style>body{ background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }</style>\n";')
            client:send(buf);
            
            conn:on("sent",function(conn)
                srv:close()	-- Even tried with client:close()
            end)
        end)
    end)
    
end