nodemcu/nodemcu-firmware #1836: TIME_WAIT sockets issues

djphoenix, 2017-03-06:
Now we have two (*or more? Point me if I miss something*) critical issues related to sockets kept in TIME_WAIT state:
- Cannot open new `server` instance when any socket in `TIME_WAIT` state with matched target port is present
- "Memory leak" (actually high consumption that rolls back after time)

I think we have a several ways to go (or maybe mix them together):
- [ ] Reduce `TIME_WAIT` state timeout for socket, so them will be closed earlier
- [ ] Kill blocking `TIME_WAIT` sockets when calling `listen` method (maybe it will require some LWIP code patching)
- [ ] Kill `TIME_WAIT` sockets somewhere in Lua GC (I not so familiar with lua-level, anyone can explain this?)

Objections? I think we should resolve this ASAP (**before** upcoming master drop).
/cc @marcelstoer @devsaurus

luismfonseca, 2017-03-06:
>  Kill blocking TIME_WAIT sockets when calling listen method (maybe it will require some LWIP code patching)

I suppose you can do this without code patching lwIP. By changing the net module and killing off `TIME_WAIT` sockets when a new connection is about to be made. Although, that means each module that uses lwIP needs to do the same thing. This is were something espconn could be of use - an intermediate layer to make lwIP more tailored to the nodemcu/esp8266 use case and capabilities :o.

djphoenix, 2017-03-06:
> Although, that means each module that uses lwIP needs to do the same thing
Yes and no - this is required feature only for TCP server sockets, TCP clients and UDP sockets haven't this requirement.

djphoenix, 2017-03-06:
So there are first "good news everyone" - I managed `SO_REUSEADDR` to work, and its the simpliest patch I wrote for nodemcu ever. *facepalm*
See 06c8a47f7d9bc709b8c50ff16984b5ae23905f5e

I will work on it some more and prepare PR.

luismfonseca, 2017-03-06:
Oh cool!

Is there any known downside for using `SO_REUSEADDR`?

djphoenix, 2017-03-06:
Not sure but now I can't reproduce memory leaks... So, *what?* o_O
@luismfonseca you have initially caught leakage, so, can I ask you to try it again and (if it reproduced) get a sample code with client-side reference?

djphoenix, 2017-03-07:
And it fixed. `TIME_WAIT` sockets are being free after `2*TCP_MSL` inactivity time frame, so for 60s MSL heap stability time is 120s. I reduced it to 5s, and stable time now is 10s. I think it is enough (unless anyone try to DDoS someone by TCP from ESP).
See 8c8947aeb35f3c249c741bd6f4cf1e73be844b87

georeb, 2017-03-08:
Will this change be available in the DEV branch soon?

djphoenix, 2017-03-08:
@georeb I promise, track #1838 for progress :)

pjsg, 2017-03-09:
One approach to dealing with time_wait is to reset the connection -- I don't know if this is possible with lwip.  (this is actually the trick that Internet Explorer uses on windows)

What happened before in the old sdk?

PR #1838 (2017-03-17): TIME_WAIT sockets fixes
Fixes #1836 (at most)
closed 2018-02-24 by marcelstoer