nodemcu/nodemcu-firmware #1841: Duplicate lua examples for DS18b20

karrots, 2017-03-07:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
One library for DS18b20 in code base.

### Actual behavior
New async library and examples in `lua_modules\ds18b20` as well as two basically identical old style example files in `lua_examples`. 

### NodeMCU version
Current `dev` be2a9df1b0fafa6b8822c24faf2a6ee5f334c676

8<------------------------ END BUG REPORT -------------------------------------

marcelstoer, 2017-03-09:
Are you willing and motivated to help us clean up here? Proposal:
- Merge https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/ds18b20/ds18b20-example.lua and https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/ds18b20/ds18b20-web.lua into a single up-to-date example
- Move that one to https://github.com/nodemcu/nodemcu-firmware/tree/master/lua_examples
- Delete the existing DS18b20 examples there
- Rename ds18b20.EN.md to README.md so GitHub would display that by default
- Correct that documentation if necessary

karrots, 2017-03-10:
Happy to create a PR. Mostly started the issue to create conversation. The issue isn't the code in `lua_modules` it's the old code in `lua_examples`.

I'll create a PR to do the following:

1. Rename `ds18b20.EN.md` to `README.md` so GitHub would display that by default.
2. Remove old examples in `lua_examples`.
3. Convert `toBase64` that I missed earlier to `toHex`
4. Fix formatting and spelling issues in `README.md`

You mention splitting the examples from the actual module. It seems like other Lua modules have examples in the folder while the examples folder is where general examples go.

PR #1851 (2017-03-10): Updated DS18b20 examples
Fixes #1841

- [ X ] This PR is for the `dev` branch rather than for `master`.
- [ X ] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ X ] I have thoroughly tested my contribution.
- [  ] The code changes are reflected in the documentation at `docs/en/*`.

--
- Removed old non-async examples from `lua-examples`.
- Renamed `ds18b20.EN.md` to `README.md`
- Changed remaining `toBase64` calls to the more standard `toHex` call.
- Fixed some spelling [... cut]
closed 2017-03-10 by marcelstoer