nodemcu/nodemcu-firmware #1842: ESP32: 1-Wire Module

LDmitryL, 2017-03-07:
## Missing feature
For ESP32 really need the 1-Wire Module to read data from the sensor AM2301 and DS18B20

### Justification
Without him is no life

devsaurus, 2017-03-10:
I'd look into this next.

API-wise I see a couple of questions:

- [ ] Move to object oriented style for buses?
    Existing API identifies a bus by its pin, create a bus object instead:
    - `bus = ow.bus(pin)`
    - `bus:<bus methods>`
- [ ] Move to object oriented style for slaves?
    - `owslave = ow.search()`
    - `owslave:read*|write*`
- [ ] Asynchronous mode with callbacks for slave reading and writing?
    Either with existing flat API or oo-API: `ow.read(cb_fn(data))`

On the hardware level I'd like to avoid bit-banging with delay loops. The Remote Control hardware seems to come in handy here. Needs some investigation.

karrots, 2017-03-11:
What about broadcast commands like start temp conversion on the ds18b20.

devsaurus, 2017-03-12:
Hm, good point. Either have a "broadcast slave" object or cover the functions with the bus itself.

Still thinking about what would be the benefit of an oo-API? Does it enable things that can't be achieved with flat API?

devsaurus, 2017-03-14:
I for myself didn't come up with good arguments pro oo-API.
Furthermore, I have a proof-of-concept based on RMT using the flat API. Will continue this path.

devsaurus, 2017-03-17:
Initial `ow` module is now in `dev-esp32` branch including [API docs](http://nodemcu.readthedocs.io/en/dev-esp32/en/modules/ow/).
Feel free to test and feed back results.

LDmitryL, 2017-03-23:
@devsaurus I'm trying to use the module to `1-wire` . Wrote here such script in the file ADrvOW.lua:

> local  drv_OW = {}
function drv_OW.read(port,var)
--local address,present,data,crc,t
    while true do
        print("Read port=",port,",var=",var,"address=",Tadr[var]);
--        address=0;
--        ow.reset(port);
--        ow.select(port, address);
--        ow.write(port, 0x44, 1);
--        present = ow.reset(port);
--        ow.select(port, address);
--        ow.write(port,0xBE,1);
--        data = string.char(ow.read(port));
--        for i = 1, 8 do data = data .. string.char(ow.read(port));  end
--        crc = ow.crc8(string.sub(data,1,8));
        print(data:byte(1,9));
        if crc == data:byte(9) then 
           t = ((data:byte(1) + data:byte(2) * 256) * 625)/ 10000;
           print("Temperature="..t.."  C")
           Tvar[var]=t;
        end       
       coroutine.yield()
    end
end
>return drv_OW

And trying to download it operator `DRV=require("ADrvOW");`
If it is done with this file, then all is well. But if you remove the comment. At least one. It reboots NodeMCU. For example from this line 
`          ow.reset(port);`
Here's what happens:
> DRV=require("ADrvOW");
***ERROR*** A stack overflow in task main has been detected.
abort() was called at PC 0x40085fb4
Guru Meditation Error: Core  0 panic'ed (abort)
Backtrace: 0x400849c7:0x3ffb7760 0x4011bb1c:0x3ffb7780 0x40083430:0x3ffb77a0 0x40084feb:0x3ffb77c0 0x4008507c:0x3ffb77e0 0x400d1689:0x3ffb7820 0x400d1716:0x3ffb7840 0x40082171:0x3ffb7860 0x40081bc5:0x3ffb7880 0x40081e34:0x3ffb78a0 0x40126c51:0x3ffb78f0 0x40126d4a:0x3ffb7920 0x4010e948:0x3ffb7950 0x40112600:0x3ffb7970 0x4011181c:0x3ffb79a0 0x4010f2b4:0x3ffb79e0 0x4010eb18:0x3ffb7a10 0x40122488:0x3ffb7a30 0x4012bbea:0x3ffb7a50 0x401255f7:0x3ffb7a80 0x401256fe:0x3ffb7ab0 0x4011f1fe:0x3ffb7ad0 0x4011ff58:0x3ffb7af0 0x40120764:0x3ffb7b60 0x40120795:0x3ffb7b80 0x4011fa25:0x3ffb7ba0 0x40120764:0x3ffb7c10 0x4011fad1:0x3ffb7c30 0x40120088:0x3ffb7ca0 0x4011fda2:0x3ffb7df0 0x401208be:0x3ffb7e60 0x4011bd80:0x3ffb7ff0 0x4011bb3c:0x3ffb8010 0x4011c3b5:0x3ffb8090 0x4011c42d:0x3ffb80c0 0x4011b8a0:0x3ffb8100 0x40122d64:0x3ffb8140 0x40129cb3:0x3ffb81f0 0x4011c1f6:0x3ffb8210 0x4011c2c4:0x3ffb8250 0x4011b808:0x3ffb8270 0x40129933:0x3ffb8290 0x4011c1f6:0x3ffb82b0 0x4011dd00:0x3ffb82f0 0x4011c2cd:0x3ffb8340 0x4011ad78:0x3ffb8360 0x4011bb3c:0x3ffb8380 0x4011c3b5:0x3ffb8400 0x4011b849:0x3ffb8430 0x4011e86b:0x3ffb8460 0x4011ebfe:0x3ffb8480
>Rebooting...

I don't think this is happening because of the module `1-Wire.` Because if you remove the `--` from the line `local address,present,data,crc,t` is the same. Just a mistake I noticed when you download this code

devsaurus, 2017-03-23:
Try to increase the main task stack, maybe double it to 8192 bytes.

It can be set with menuconfig, see https://github.com/devsaurus/nodemcu-firmware/blob/dev-esp32_build/docs/en/build.md#stack-size

LDmitryL, 2017-03-23:
@devsaurus Thank you. It seems to have helped. Booted without error.

devsaurus, 2017-03-27:
@jmattsson the stack for main task is currently set to 4096 by default. Should I just increase it to whichever value that appears to fix such crashes?

jmattsson, 2017-03-28:
I guess we don't have much choice here, much as I don't like throwing memory at the stack...

IIRC there's a relatively easy way to provide custom default-values for the IDF config (a target creating the `.config` before the Kconfig stuff gets run); I think I saw one of the examples using it.

devsaurus, 2017-03-28:
Excellent, thanks for the hint!
Found the details at http://esp-idf.readthedocs.io/en/latest/build-system.html#custom-sdkconfig-defaults and increased stack size to 6&nbsp;kByte for the time being.

devsaurus, 2017-03-29:
No further open items, considering this as done.

closed 2017-03-29 by devsaurus