nodemcu/nodemcu-firmware #1845: Using am2320 on existing I²C bus

v1993, 2017-03-08:
### Missing feature
Work with asm2320 on already initialized I²C bus with another slave devices.

### Justification
I want to connect asm2320 and I²C display over one bus

### Workarounds
Just comment i2c setup in asm2320.c

devsaurus, 2017-03-08:
Agreed, the inherent i2c setup potentially conflicts with multiple slaves on the bus.

FrankX0, 2017-03-08:
I also agree, but keep in mind that to keep consistency, all other i2c slaves (ADXL345, BME280, etc.) also need similar modifications.

v1993, 2017-03-09:
I think, we need to split `setup` function into two: new `begin`, that initializes sensor on exiting bus and old `setup`, that setups new I²C bus and just runs `begin` (only for compatibility).

devsaurus, 2017-03-27:
As @FrankX0 pointed out, a grep for `platform_i2c_setup` lists the following modules. IMO all of them need the same change to remove global I2C bus initialization:
- am2320.c, `am2320.init()`
- adxl345.c, `adxl345.init()`
- bme280.c, `bme270.init()`
- bmp085.c, `bmp085.init()`
- hmc5883l.c, `hmc5883l.init()`
- l3g4200d.c, `l3g4200d.init()`

Picking up @v1993's proposal, I see a two-step approach.
1. Refactor `xyz.init()` into I2C bus setup and `xyz.setup()` for the bare module hardware. Has the same overall behavior as before, but emits a deprecation warning via #1538.
    Users should switch to `xyz.setup()` but can keep their scripts untouched for the time being.
2. Remove `xyz.init()` after next drop.
    `xyz.setup()` is the only initialization / setup function remaining.

Is this the right way to go?

FrankX0, 2017-03-27:
Sounds good to me!
Let's not forget to modify the new Hdc1080 module from initial implementation.

PR #1887 (2017-05-21): Deprecate init() in favor of setup(), removing inherent i2c configuration: adxl345, am2320, bme280, bmp085, hmc5883l, l3g4200d
Fixes #1845.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

Having the I2C configuration included in the modules' init functions is an issue for a shared b [... cut]
closed 2017-05-21 by marcelstoer