nodemcu/nodemcu-firmware #1849: HTTP Module "Receive Event" Payload Size Limit

NicolSpies, 2017-03-09:
### Expected behavior

When using **tls.createConnection** with a HTTP GET it should be possible to receive a medium sized file (2kb) from a server.

### Actual behavior

It is only possible to receive a file up to about 627 bytes. With the HTTP headers of about 800 bytes the size of the received payload is then about **1458 bytes which appears to be the maximum allowed payload** size.

**Larger files** result that the "**on receive" event is triggered twice** . When this occurs the total payload is either not received or is over written resulting that the file downloaded is not usable due to missing data.

### Test code

```Lua
local host = "raw.githubusercontent.com"
local path = "/NicolSpies/sandboxie/master/test10.lua"
local srv = tls.createConnection(net.TCP, 0)

srv:on("receive", function(sck, c) 
    print(#c.."\n")
    print(c)
    srv:close()
end)    

srv:on("connection", function(sck, c)
    print("connected")
    sck:send("GET " .. path .. " HTTP/1.1\r\nHost: " .. host .. "\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n")
end)
   
srv:connect(443, host)
```
### NodeMCU version

NodeMCU custom build by frightanic.com
	branch: dev
	commit: 466c03d90fb0618488e9e790200777ed52c83ad2
	SSL: true
	modules: adc,bit,cjson,coap,cron,file,gpio,http,i2c,mdns,mqtt,net,node,rtcfifo,rtcmem,rtctime,sntp,struct,tmr,uart,websocket,wifi,tls
 build 	built on: 2017-03-08 06:51
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Hardware

ESP12-e on a nodemcu-devkit

### Missing feature

@marcelstoer I suspect the maximum payload size might have been reduced last year to eliminate HTTP & SSL issues. It might also be related to the issue reported in #1821 where the CJSON decode is limited to 250 about 250 characters.

### Justification

From example code in GitHub it appears as if this limitation did not exist in the past as larger.

### Workarounds

Using http.get as used in #1707 does not work either and fails with the following information: HTTP client: Disconnected with error: 8

djphoenix, 2017-03-09:
I suspect this is not a bug. You will catch `receive` callback for each **frame** of data, which limited to TCP MTU and more (e.g. WiFi fragments). Because TLS there is some overhead for each packet. So you should implement stream defragmentation in lua level (or wait for #1707 to resolve).

NicolSpies, 2017-03-09:
@djphoenix , thanks for feedback. If I understand everything is fine if payload size is <= one frame of data. Larger payload results in 2 frames and current net.tls module does not have way to combine multiple frames into one payload.

NicolSpies, 2017-03-09:
@djphoenix , not a bug. The multiple receive event trigger is next frame received. Fixed with the following: 

    local buffer=nil

    srv:on("receive", function(sck, c) 
        if buffer==nil then
            buffer=c
        else
            buffer=buffer..c
        end
    end)

marcelstoer, 2017-03-09:
Yes, that ain't a real bug. I remembered an open issue way down on the list and found #960.

One could of course argue that this be a missing feature but one could also argue that it be perfectly fine that application developers need to handle that manually. What if you accidentally attempt to download a 2MB resource...

What would be helpful though is if this were properly documented.

NicolSpies, 2017-03-09:
I think an example with my code will explain how to defragment the received frames keeping in mind the buffer variable will grow and eat into the available heap.

Using #buffer, tls.socket.hold() and an appended file write could possible be used to throttle flow and "build/save" a large resource.

djphoenix, 2017-03-09:
@marcelstoer maybe there is a way to get `:stream(<file>)` method for HTTP request for "download" requests. Anyway I think #1707 is priority for this issue (and for `net`/`tls` sockets `receive` event per-frame is OK).

marcelstoer, 2017-03-09:
Documentation fixed.

closed 2017-03-09 by marcelstoer

NicolSpies, 2017-03-09:
@marcelstoer nice :+1:

commit e8d5a059 (2017-03-09): Document that the socket receive event is fired for every frame, fixes #1849