nodemcu/nodemcu-firmware #1850: New http server

israellot, 2017-03-09:
I'm working on a new http server for nodemcu. 
It has a dynamic module concept, pipelined, and right now it has a working filesystem module with path and dir mapping, index searching, cors and some other stuff. It uses the nodejs http parser and can stream data easily. 
As easy as 
```
server = httpserver.new()        
        server:use_filesystem('/static','/fileprefix')
        server:listen(80) 
```

I am also trying to separate the engine layer from the esp layer so it can be reused elsewhere. 

Anyone wants to join?

https://github.com/israellot/nodemcu-firmware/tree/httpserver/app/httpserver

djphoenix, 2017-03-09:
@israellot you're welcome :)
First of all, what about customisation of your module? For example, can user disable "index search" module for his build to keep some ROM?
Second thing, what about memory consumption? For IC, we should keep it as low as possible...
Anyway you're doing good things, welcome again 👍

israellot, 2017-03-09:
When idle memory consumption is very low. Just some pointers and config data. Processing requests can consume memory, but only if needed. Request parsing is done in a copyless manner, each module can configure which data it needs from the request in a way that the engine will only save the headers it was requested to save. When serving files from flash, the biggest tradeoff that has to be made is the chunk size of each read-read turn. Smaller chunks will lead to less memory consumed but also to slower deliveries. That's configurable. 
There's still a lot to be done, it's early stage yet, but looks promising.

pjsg, 2017-03-10:
Before you go too far, I think it sensible to write the documentation page. This way, we have have a discussion about the API before too much code gets written. 

For example, I'm interested in how scripts are attached to paths in the URL. How are these invoked? Can they generate result data in chunks? The new `sjson` module is designed to be able to encode a lua object into json -- but in chunks that can get sent individually. This reduces the overall memory requirement significantly.

israellot, 2017-03-10:
You are right @pjsg. I'll take some time to document what has been done already. 
I'm still architecting the api module, the one that will allow users to attach scripts to urls, possibly with pattern matching.
From my experience with the chunking json data, the biggest problem is that you need to have the whole object ready to only then parse it to json. So it's a double memory requirement, specially if your json values are long strings, in such a way that the json encoding is not a huge overhead compared to the size of the source object. SJson looks promising on reducing the requirements on output, I'll look into it deeper.

fikin, 2017-03-21:
@israellot : i'm looking for something like this one, with exact same properties (proper http request handling, simple callbacks on url, streaming json body to lua converter). let me know if you want contribution in some specific way.

from practical pov though, i see benefits if nodejs http-parser exposed to lua on its own, with option to integrate to existing net module. this will bring immediate benefits without the need to adopt completely new server infra. any opinions about not doing it?

closed 2018-02-24 by marcelstoer

marcoskirsch, 2018-02-27:
Did something come out of this?