nodemcu/nodemcu-firmware #1854: Memory leak

jankop2, 2017-03-10:
I'm clueless. Here is a brief functional fragment off http server. It is small and simple.  Despite the best efforts can I not prevent fast systematic loss of memory and system goes in a short time to the collapse. Can you tell me please why?

```
YourSSID ="********"
YourPassw="********"
wifi.setmode(wifi.STATION)
wifi.sta.config(YourSSID,YourPassw)
wifi.sta.connect()
tmr.alarm(0,1000,1,
    function()
        staip,_,_=wifi.sta.getip()
        print("Wait for IP...")
        if staip~=nil
        then
            tmr.stop(0)
            print("** server IP :",staip)
        end
    end)
srv=net.createServer(net.TCP,60)
srv:listen(80,function(conn)
	local buff=" "
    conn:on("receive",function(client,payload)
            local function SendData()
                buff=
				'<!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN">\
				<html><head>\
				<meta content="text/html; charset=utf-8">\
				<title>Dual-switch</title></head>\
				<body style="color: brown;background:#ffe4c4">\
				<h1>JankoP - Factory</h1><br>\
				<form action="/" method="post">\
				<input type="submit" name="REFRESH" value="REFRESH"><br></h2><br>\
				<br>Heap : '..node.heap()..'</form></body></html>'
                lenght= #buff
                buff =
				'HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\Content-Length: '.. lenght ..
				'\r\nCache-Control: no-cache\r\n\Pragma: no-cache\r\n\Connection: close\r\n\r\n'..buff
                client:send(buff)
                conn:on("sent", function(client) client:close() buff=nil collectgarbage() end)
            end
            if string.find(payload,"POST / HTTP/%d%.%d\r\n.+\r\n\r\n(.+=.+)")~=nil
            then
                SendData()
            elseif string.find(payload,"GET / HTTP/%d%.%d\r\n.+\r\n\r\n$")~=nil
			then
                SendData()
            end
			payload=nil
			collectgarbage()
    end)
end)
```
NodeMCU custom build by frightanic.com
branch: master
commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
SSL: false
modules: file,gpio,net,node,tmr,uart,wifi
build 	built on: 2017-03-09 19:58
powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
 
I use standard esp8266-12F with 4MB flash memory

NicolSpies, 2017-03-10:
timer usage looks suspect, try `tmr.alarm(0,1000, tmr.ALARM_SINGLE, function...`

NicolSpies, 2017-03-10:
Timer loop fires every 1000ms while waiting for wifi ip  and in every loop all the http server code is executed again killing heap. Split into two functions where the only the wifi function is in the timer loop and calls the http server function once an IP is obtained.

jankop2, 2017-03-10:
No, it is not right. Server is start only one. It is Lua, not Arduino.

marcelstoer, 2017-03-10:
See #1010, this contains nothing about memory constraints that we don't already have issues for AFAICS, looks like Lua developer support.

closed 2017-03-10 by marcelstoer

jankop2, 2017-03-10:
Yes, I understand. Lua is dead. It has lots of features, but they can not use, because even a program that uses only 1kB memory collapses after a short while. I'm sorry, because I with esp8266 Lua he spent a lot of time.