nodemcu/nodemcu-firmware #1855: MQTT client hang connecting on hairpinning / NAT server from inside

sivix, 2017-03-11:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
I've an MQTT server with NAT and hairpinning configured.
I'm able to connect from outside and inside using different MQTT Clients (i.e MQTT.fx).
Configuring the MQTT external IP ADDRESS, esp8266 should be able to connect from inside network 192.168.0.x.

### Actual behavior
if ESP8266 is connecting from internet (outside) MQTT client get connected, from Inside I get a connected = true status, but when I subscribe I got an error, "Not connected".
NO Failure code returned, looks like MQTT Client seems to hang.

### Test code

mqttServer = "public_ip" / "internal_ip"
mqttTopic = "topic/"
m = mqtt.Client("NODE",15,nil,nil,1)
m:on("connect",function(c) print("CONNECT") end)
m:on("offline",function(c) print("OFFLINE") end)
m:on("message",function(c,t,m) print("MESSAGE",t,m) end)
m:connect(mqttServer,1883,0,0)
m:subscribe(mqttTopic,2)

public_ip = ext_NAT:1883
internal_ip = 192.168.1.x:1883

### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
NodeMCU custom build by frightanic.com
	branch: master
	commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
	SSL: true
	modules: adc,enduser_setup,file,gpio,mqtt,net,node,pwm,tmr,uart,wifi,ws2812,tls
 build 	built on: 2017-02-23 12:48
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

ESP8266 AMICA, and simple ESP8266-12e.

8<------------------------ END BUG REPORT -------------------------------------

djphoenix, 2017-03-11:
Can I ask you to try the `dev` build? Also, I think you should move `:subscribe` call to `connect` callback handler.

_I suppose that you was configured your router/firewall right for port forwarding, so I'll not ask you about configuration check, but if **you** unsure - check it in your way_

sivix, 2017-03-11:
Hi @djphoenix, Yes, I'm 100% sure of router/firewall since I've test it also from other clients from internal network for example with my Mac using MQTT.fx I can connect to public address, even from iphone attached to the internal wireless. from ESPP internal network I tried the Net.TCP connect on public IP on a different port and I have no problems.

about the :subscribe call you are right works better, but I have the same problem, since it looks like MQTT:connect, doesn't call the success or fail function... just simply hangs.

I'll try the dev firmware and let you know, asap.

sivix, 2017-03-11:
Hi,  same behaviour.

this is the code tested:

mqttServer = "xx.xx.xx.xx"
mqttTopic = "smartButtons/"
m = mqtt.Client("NODE",15,nil,nil,1)
m:on("connect",function(c) print("CONNECT");m:subscribe(mqttTopic,2); end)
m:on("offline",function(c) print("OFFLINE") end)
m:on("message",function(c,t,m) print("MESSAGE",t,m) end)
m:connect(mqttServer,1883,0,0)

the oputput is this one:
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 725feade27813d0f5a323d504cc259bc24ec38e4
	SSL: true
	modules: adc,enduser_setup,file,gpio,mqtt,net,node,pwm,tmr,uart,wifi,ws2812,tls
 build 	built on: 2017-03-11 15:29
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
System voltage (mV):	3528
Starting SmartButton
> (1) Waiting for IP...
> (2) Waiting for IP...
192.168.1.8	255.255.255.0	192.168.1.254
CONNECT
OFFLINE

--
Funny is that print "CONNECT" but suddenly goes offline

if I test it with internal IP:

NodeMCU custom build by frightanic.com
	branch: dev
	commit: 725feade27813d0f5a323d504cc259bc24ec38e4
	SSL: true
	modules: adc,enduser_setup,file,gpio,mqtt,net,node,pwm,tmr,uart,wifi,ws2812,tls
 build 	built on: 2017-03-11 15:29
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
System voltage (mV):	3531
Starting SmartButton
> (1) Waiting for IP...
> (2) Waiting for IP...
192.168.1.8	255.255.255.0	192.168.1.254
CONNECT
MESSAGE	smartButtons/	ciao

the mqtt is message successfully received (is a retained, Q0)

djphoenix, 2017-03-11:
I have tested connection with port-forwarding in my network, so I can confirm that it works (for local, local with external-ip, and even for remote connections via phone hotspot). So I suspect problems in forwarding or firewalling anyway. Can you run Wireshark on PC with broker running, and reproduce issue?

sivix, 2017-03-11:
Hi, I'm running mqtt on a raspberry; 
I've attached a tcpdump capture of connection from Mac and from the ESP8266.

for the ESP-NODE-callbacks.pcap I used the following code with :connect() callbacks as suggested.

mqttServer = "external_IP"
mqttTopic = "smartButtons/"
m = mqtt.Client("NODE",15,nil,nil,1)
m:on("offline",function(c) print("OFFLINE") end)
m:on("message",function(c,t,m) print("MESSAGE",t,m) end)
m:connect(mqttServer, 1883, 0, 
        function(client) 
            print("CONNECTED") 
            m:subscribe(mqttTopic,0, function(sub) 
                print("subscribed: "..mqttTopic) 
              end)
        end, 
        function(client, reason) 
            connected = false 
            print("failed reason: "..reason) 
end)

[tcpdump.zip](https://github.com/nodemcu/nodemcu-firmware/files/836020/tcpdump.zip)

closed 2019-06-21 by stale[bot]