nodemcu/nodemcu-firmware #1856: enduser_setup: Show obtained IP address to user after connected to WiFi

tricoos, 2017-03-11:
### Missing feature
WiFi IP not shown to user in captive portal after successful connection

### Justification
The user cannot find out the IP address of the device easily except via NMAP or DHCP server logs. But the IP address is important if you have a web server running on the device as the user needs to know the IP to connect to it.

### Workarounds
Not using enduser_setup and writing it from scratch in LUA.

FrankX0, 2017-03-11:
This feature used to be in there ( #1313).
I'll see what I can do.

pjsg, 2017-03-11:
Another approach is to use mdns to advertise the name of the device and then you can connect by name.

FrankX0, 2017-03-11:
I'm also using mdns for my applications.
But note that this works in Windows and iOS, but not (AFAIK) in Android.

pjsg, 2017-03-11:
Argh! I just starred https://bugs.chromium.org/p/chromium/issues/detail?id=405925

I did install the 'Pea Finder' app (https://play.google.com/store/apps/details?id=com.ersteheimat.peafinder) which shows the local mdns servers which support web access and then allows you to click on them.

FrankX0, 2017-03-11:
I also just installed ' Pea Finder', which indeed finds device with mdns enabled.
But I don't think this is always a convenient way to first having to install this app to find the device in question.

pjsg, 2017-03-12:
I think that the best approach is to get Chrome to support mdns on Android.... But I don't know how to get them to do it....

tricoos, 2017-03-17:
Well, just including the mdns module to connect to the module once and then never use it again when the IP address is readily available is something I don't want to do because of memory reasons. In addition, to be able to use mDNS you need to have Bonjour/Avahi installed on the host system. But then again even if I would use mDNS how do I show the user the hostname he's supposed to connect to? I can't, right? So I don't have any advantage over showing the IP address that's readily available.
I have multiple devices of the same type in the same network so knowing the IP or the hostname is required to know which of them is the new one I have to connect to to configure/control it.
And mDNS is no solution as I don't have any memory left and was able to get the code running with some heavy optimizations to save RAM. So showing the IP address that's readily available is the only solution for me with this implementation.

FrankX0, 2017-03-17:
I agree that mDNS is not always the solution.
Please have a look at the referenced pull request.
After connecting to the AP it will show the assigned IP address.

FrankX0, 2017-03-20:
Propose to close: with the merge the request to show the IP address has been implemented.

PR #1867 (2017-03-18): Show IP address after enduser_setup
Fixes #1856.

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/en/*`.

After `enduser_setup` has successfully connected to the AP, the assigned IP address will be shown instead of the device_id.
closed 2017-03-21 by marcelstoer