nodemcu/nodemcu-firmware #1857: wifi: wifi.sta.getip should return DNS servers as well

tricoos, 2017-03-11:
### Missing feature
wifi.sta.getip() returns IP, netmask and gateway IP address, but not the DNS server(s) used, although they would've been obtained in station mode anyway.

### Justification
In my opinion DNS resolution is part of a normal station so I should be able to get this value from the wifi module.

### Workarounds
net.dns.getdnsserver()

djphoenix, 2017-03-16:
I think no. IP addresses (with routing table) are related to **interface**. DNS server addresses are related to **system** (even if they're received with DHCP from WiFi router). So right place for them in `net` module.

tricoos, 2017-03-17:
DNS servers are related (on normal systems) to the interface, not the system . There are no system DNS servers in general AFAIK. On a more complex system (Windows, Linux etc.) you can for example be connected to multiple networks via multiple interfaces with each interface having its set of nameservers. These might be queried one after the other for name resolutions but even though they are still related to the corresponding interface.

marcelstoer, 2017-03-17:
> ...a more complex system ... be connected to multiple networks via multiple interfaces...

True, but IMO it ain't helpful to compare IoT/embedded devices with those. I think the initial creators of this firmware weren't so wrong to place DNS info in `net.dns.getdnsserver()`.

marcelstoer, 2017-03-28:
> IP addresses are related to interface [...] DNS server addresses are related to system

True or false, whatever it is... on ESP8266 et.al. with only one interface present it doesn't really make a difference.

With none of the current maintainers coming forward in support of this request I move to close it.

closed 2017-03-28 by marcelstoer