nodemcu/nodemcu-firmware #1858: Reduce the number of GCs by trading off a bit of memory

pjsg, 2017-03-11:
### Justification
A good percentage of the time in a typical nodemcu/lua app is spent performing garbage collections. A significant perfomance improvement can be got by reducing the number of GC runs.

### Workarounds
You can change the egc mode to be not as aggressive, but then you run into problems with the C code that allocates memory and which doesn't handle failure gracefully.

### Proposal
This new approach keeps a few block of recently freed memory in a list that can be handed back quickly if needed. Some experiments show that (for the small blocks -- up to 40 bytes), maybe 80% of the allocations can be handled by keeping 2kB of total blocks lying around. 

There will be a #define that controls the amount of memory that can be saved.

TerryE, 2017-03-17:
Philip, I've looked at this one in the past; you have to tread carefully here.  The GC is core Lua RTS code and tweaking it can have unintended consequences.  Certainly the current build use `EGC_ALWAYS` which just gives crazy overhead for no good reason, but if you want to keep a few Kb free for the C code that doesn't use the Lua allocator, then why not just set

```C
legc_set_mode( L, EGC_ON_MEM_LIMIT, 4096 )
```
which does this?

And incidentally except in the rare situations where the C application can take programmatically sensible actions on memory exhaustion, it is always safer to use the Lua allocator.

jmattsson, 2017-05-04:
Anecdote: We've been using the `EGC_ON_MEM_LIMIT` approach in production for some time, without any issues.

TerryE, 2017-05-04:
If we deprecate the use of bare malloc for anything other than startup initialisation and stick to the lua allocator then the mem limit approach is very robust, IMO, so my recommendation is to close this.

HHHartmann, 2017-07-24:
Might it be reasonable to change the build default to use EGC_ON_MEM_LIMIT then?
Helps  most users not aware of GC at all and the few which know have their own settings anyway.
Just my 2c

closed 2019-08-04 by stale[bot]