nodemcu/nodemcu-firmware #1861: Add a sleep/dsleep chapter to Lua Developer FAQ

marcelstoer, 2017-03-12:
@TerryE's FAQ has a nice chapter about [techniques for reducing RAM and SPIFFS footprint](http://nodemcu.readthedocs.io/en/latest/en/lua-developer-faq/#techniques-for-reducing-ram-and-spiffs-footprint). I believe it'd very useful to add a related chapter about "How to reduce power consumption and how to sleep/dsleep".

We have documentation on [`node.sleep`](http://nodemcu.readthedocs.io/en/latest/en/modules/node/#nodedsleep), [`wifi.nullmodesleep`](http://nodemcu.readthedocs.io/en/latest/en/modules/wifi/#wifinullmodesleep), [`rtctime.dsleep`](http://nodemcu.readthedocs.io/en/latest/en/modules/rtctime/#rtctimedsleep), and [`rtcfio.sleep_until_sample`](http://nodemcu.readthedocs.io/en/latest/en/modules/rtcfifo/#rtcfifodsleep_until_sample). What we're missing is the bigger picture. When to use which? Which function is intended for which use case? What is the best strategy for (quasi-)autonomous, battery/solar-powered devices? What are the gotchas? It'd also be nice to document the required wiring graphically (linking to `node.sleep`).

@eyaleb or @jmattsson would you have time to put together a draft (assuming you agree with my request)?

jmattsson, 2017-03-12:
Good idea, but at the moment I'm still somewhat overloaded :/  While I've started to catch up on things a wee bit, I've still got something like 250 outstanding github notifications to make my way through.

TerryE, 2017-03-12:
I've just finished doing the plumbing and central heating system for my new house.  And then I want to crack on with the Lua 5.3 implementation which kinda got stalled with the house work.  Sorry.  In terms of the FAQ, there's bits that really need updating like once we've got the early memory leaks out of the way, using UpVals is solid, fast and efficient, but there are gotcha's which need explained for the beginners.

So what I am saying is that this is a great idea, but not one that I've got time for at the moment.

flip111, 2017-03-26:
@TerryE i was going to open a new issue, but i saw you mentioning lua 5.3. Is Lua 5.3 support in the works? That be pretty sweet, i just checked the lua homepage and 5.1.* seems pretty old.

TerryE, 2017-03-26:
@flip111, see #1661 for this discussion.

TerryE, 2017-04-05:
@marcelstoer Marcel,  I need to add this and I will include this in the next push of my draft.  Thanks :smile:

TerryE, 2017-04-05:
> #1861 and #1862. I wasn't going to mention them but since you explicitly asked for such feedback I saw no point in withholding them.

Marcel, definitely #1861 belongs in here.  #1862 belong in the Extension Developers FAQ, though there is a parallel  one about how Lua developers have debugger support.  Even if we ported the rest of the `debugger` library it is really only an set of RTS hooks so an IDE can integrate with Lua.  I am not sure the best way to progress this one.

closed 2019-06-21 by stale[bot]