nodemcu/nodemcu-firmware #1862: Add a "How To Debug" chapter to Extension Developer FAQ

marcelstoer, 2017-03-12:
We currently have [module documentation for `gdbstub`](http://nodemcu.readthedocs.io/en/latest/en/modules/gdbstub/). It's quite a bit more than a simple API documentation but not enough to qualify as a real "How to". IMO it would be helpful to have recipe-style instructions in the [extension developer FAQ](http://nodemcu.readthedocs.io/en/latest/en/extn-developer-faq/)s for people who have never worked with GDB. Such a chapter would also prove to be useful in bug processing. We could send people there with a clear conscience and ask them to do some of the debugging themselves.

TerryE, 2017-03-13:
Marcel, one thing that I keep asking myself is:  why we don't implement the standard Lua debug library?  It is really quit useful in bug hunting end-user Lua apps.

As to the **gdb** stub, I personally feel that we should assume that C developers will already be familiar with **gdb** (and if they aren't, then there are already a lot of useful tutorials out there).  What we could do with is a more focused explanation of how to do host / target debugging using the **gddbstub** and its strengths and limitations.

marcelstoer, 2017-03-13:
Nice feedback Terry, thanks.

> why we don't implement the standard Lua debug library? It is really quit useful in bug hunting end-user Lua apps.

I'm sure we wouldn't reject a PR if someone provided an implementation, would we?

> As to the gdb stub, I personally feel that we should assume that C developers will already be familiar with gdb

I couldn't tell. What I do know though is that in 17+ years of Java development I never interacted with the Java debugger directly - the complexity is all nicely tucked away in the IDEs.

> and if they aren't, then there are already a lot of useful tutorials out there

That must be true but nothing stops us from adding a reference to your favourite tutorial.

> What we could do with is a more focused explanation of how to do host / target debugging using the gddbstub and its strengths and limitations.

That's essentially what I had in mind. Philip's documentation contains aspects of that already.

I created this isssue here because I personally would profit from such an FAQ chapter. _Reading_ and _debugging_ someone else's code in a language that you are not/no longer/not yet familiar with is the first step into understanding. That's my status with this firmware code.

TerryE, 2017-10-21:
@marcelstoer @jmattsson @pjsg after my experiences on debugging this #2068, I have decided that we definitely need this.  We also need a more developer friendly way of debugging and testing C library code.

I've got the GDB stub working well -- albeit with nasty quirks -- but to do this I've had to patch the Espressif library, the UART driver, and various Lua modules.  The main catch-22 is that if you want to debug at a C statement level then you've _got_ to allow the GDB stub to take control of UART0.  And this means that you _must_ lose access to the interactive Lua prompt -- which is a total pain. So I have two approaches for doing adhoc Lua testing to exercise a C library that I am debugging:

-  **Rapid file system replacement**. I have a script which allows me to update a 32Kb filesystem at a fixed location and then inject it into the test 8266.  So I can make a quick change to `init.lua` save it  and download to the ESP module and reboot into it within ~ 3 sec.    This allows me to do my Lua tests by quick replacement of the startup init file.

-  **Telnet Server**.  This just boots into my variant of the Telnet server so I can connect to it using `screen` and issue Lua commands that way, but one slight hassle is that this only redirects `stdout` and not `stderr` so output comes back to the Telnet session, but errors are reported elsewhere.

I've also had to tweak the debug and error routines so that you can redirect error messages / stderr to UART1.  So now my "on holiday" mobile rig consists of my laptop, a Wemos D1 chip with UART1 connected through a USB serial UART chip so I have 2 USB serial connections: one at  115200 for the GDB session and one at 460800

TerryE, 2017-10-21:
Sorry about the close and open -- finger trouble.  If I am honest, then I would _never_ have got this flash stuff working without a usable remote debugger.  Goodness knows how @zeroday got the first version of the Lua system bootstrapped.  I am just in awe.

I did the LCD patch by getting it working under a standard Linux environment using the patch on std Lua then moving onto NodeMCU, but for stuff like this Flash mod, Lua 5.3 or any complex H/W related libraary, then IMO you really need decent `gdb` access to develop this sort of stuff, and because of the UART0 conflict, this isn't easy.

closed 2019-10-26 by TerryE