nodemcu/nodemcu-firmware #1864: ESP32: NET module

michan18, 2017-03-15:
Hello, I was trying to implement a TCP server on ESP32 and listing via a port a Panic issue is thrown out ..This is my sample....code 

srv=net.createServer(net.TCP,0)
srv:listen(80,function(conn)
conn:on("receive",function(conn,payload) end)
print(payload)
end)

Whenever I execute the code i am getting PANIC: unprotected error in call to Lua API (attempt to call a nil value)

Now someone to take over to actually fix it or a possible solution?

FrankX0, 2017-03-16:
Shouldn't the `print(payload)` not be part of the `conn:on` callback (incorrect position of the first end)?
Like this:
```lua
srv=net.createServer(net.TCP,0)
srv:listen(80,function(conn)
  conn:on("receive",function(conn,payload)
    print(payload)
  end)
end)
```

michan18, 2017-03-16:
Thank you but the issue is on port listening ...
srv:listen... will always showing a panic issues.....only in esp 32

FrankX0, 2017-03-16:
To me it sounds there is an issue with the callback mechanism of the `listen` function.
Can you try the following:
```lua
srv=net.createServer(net.TCP,0)
srv:on("receive",function(conn,payload)
  print(payload)
end)
srv:listen(80)
```

michan18, 2017-03-16:
Yes tried it but its returning with  an error 
 attempt to call method 'on' (a nil value)

michan18, 2017-03-19:
Yes ...Thanks its fixed now........:)

FrankX0, 2017-04-27:
@michan18: can be closed I guess?

michan18, 2017-04-28:
Yes

PR #1869 (2017-03-19): ESP32: Fix event id for post_net_accept()
Fixes #1864.

Johny, please have look:
`post_net_accept()` set event id to `SENTDATA`. PR corrects this to `ACCEPT`.

I've also followed the pattern to check cb ref for `LUA_NOREF` in `laccept_cb()`. Note the relative stack reference for `lua_pushvalue()`.
closed 2017-04-29 by devsaurus