nodemcu/nodemcu-firmware #1868: It appears that spiffs is not safe if power is lost at a bad time

pjsg, 2017-03-17:
See https://github.com/pellepl/spiffs/issues/137

It appears that the solution is to runs SPIFFS_check after restarting. My concern is that this could be slow. In particular, this might cause problems in systems which go into deep sleep and are then woken up (when, presumably, the filesystem is actually in a good state). The issue is that we need to read from the filesystem to get the init.lua.

pjsg, 2017-03-22:
However, it appears that the naive solution of running spiffs_check can make things worse....

closed 2019-06-21 by stale[bot]

eastmoutain, 2019-11-09:
any solution for power loss issue?

TerryE, 2019-11-10:
> any solution for power loss issue?

Nope.  SPIFFS is not a journaling FS, so there is no transactional integrity around writes to the FS.  The read, write and erase primitives simply call the SDK API calls for these operations.  Power failure during write or erase can leave the FS in an unknown and potentially corrupt state, though the logical design attempts to mitigate these issues.  See [How does SPIFFS handle powerlosses?](https://github.com/pellepl/spiffs/wiki/FAQ#how-does-spiffs-handle-powerlosses).