nodemcu/nodemcu-firmware #1870: file.stat - memory leak?

EgorOA, 2017-03-19:
I'm using file.stat to manage the size of my log files

Code like this
```
		local stat = file.stat(curDataFileName)
		if stat and (stat.size > 8192) then
			-- do rename data file and starts new one
		end
		stat = nil

```
causes memory leak every time when called before writing to file.

firmware:
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 46f651cccb15a6a27e044c8afb0eca2e1334e8f0
	SSL: false
	modules: adc,bit,bme280,dht,enduser_setup,file,gpio,i2c,mdns,mqtt,net,node,ow,rtctime,sntp,spi,tmr,uart,ucg,wifi
 build 	built on: 2017-03-16 11:59
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)
```

EgorOA, 2017-03-19:
For me now workaround for this trouble is 
```
		local f = file.open(curDataFileName)
		if f then
			local size = f.seek('end')
			f.close()
			if (size > 8192) then
				-- do rename data file and starts new one
			end
		end
		f = nil
```
And where is no memory leak then getting file size this way.

Look at the file.stat closely, please!

devsaurus, 2017-03-19:
Can confirm the leak, will have a look.

EgorOA, 2017-03-19:
It took me long time to find this bug

PS. Good way for monitoring memory leak in alive system is to publish heap size e.g. to thingspeak
I do it something like that https://thingspeak.com/channels/230321

devsaurus, 2017-03-19:
Cool setup 👍
The leak is fixed in the related PR, thanks for reporting!

PR #1871 (2017-03-29): Fix memory leak in file_stat().
Fixes #1870.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

Function `file_stat()` missed to free the temporary stat object which caused a memory leak.
closed 2017-03-29 by marcelstoer