nodemcu/nodemcu-firmware #1872: GPIO16 Levels during Deep Sleep

NicolSpies, 2017-03-20:
### Expected behavior

**node.dsleep()** and **rtctime.dsleep()** should set GPIO16 to a HIGH level until timed out and then set the level LOW. The connection of GPIO16 to the RST pin (actve low) will then drive the RST pin low causing a wakeup due to the reset.

### Actual behavior

**The GPIO16 levels during deep sleep are inverted** resulting in a LOW level when deep sleep is activated instead of a HIGH level. This results that GPIO16 and RST cannot be connected together for the automatic wake-up as an immediate reset occurs on deep sleep activation. 

### Test code

```Lua
node.dsleep(5000000, 2)
```
### NodeMCU version

NodeMCU custom build by frightanic.com
	branch: master
	commit: b96e31477ca1e207aa1c0cdc334539b1f7d3a7f0
	SSL: true
	modules: adc,bit,cjson,coap,cron,file,gpio,http,i2c,mdns,mqtt,net,node,rtcfifo,rtcmem,rtctime,sntp,struct,tmr,uart,websocket,wifi,tls
 build 	built on: 2017-03-09 08:51
 powered by Lua 5.1.4 on SDK 2.0.0(656edbf)

### Hardware

NodeMCU Dev-Kit R2 with ESP12-E

karrots, 2017-03-21:
Sounds kind of like your devkit board is missing a pull-up resistor on the reset pin. Pretty sure the GPIOs all float during deep sleep.

NicolSpies, 2017-03-21:
Unfortunately not, I have confirmed and measured that pull-up resistor on RST exists. I have isolated GPIO16 from RST and added a pull-up resistor to GPIO16. External power is provided to the module so that the RS232 cannot interfere and confirmed the following:

- Before node.dsleep() command GPIO16 is HIGH via the pull-up resistor as expected.
- When node.dsleep() command executes the GPIO16 is immediately pulled LOW. This is not expected. It should stay HIGH until wake-up and then be pulled LOW to simulate/create the reset pulse.

eyaleb, 2017-03-21:
I have no pullup at all on my naked esp-07, just RST connected to GPIO16. In deep sleep it stays high (floating, probably with internal pullup). I am on the dev branch but the sleep/wakeup stuff is SDK functionality.

Pulling RST down should not trigger a reset, the following low-to-high transition is the actual trigger.

Anyway, what you describe should make wakeup not work for everyone, right? I do not see many "me too"s. Did dsleep work for you before?

OK, I must ask: are you sure it is GPIO16 you are connecting?

HTH

NicolSpies, 2017-03-21:
It was working all along. Me being stupid and in need of new glasses:
- sleep time in us not ms, so rebooting was wake-up cycle.
- short pulses can only be seen on a scope and not a multi-meter (duh)

I humbly apologize for wasting everybody's time

closed 2017-03-21 by NicolSpies