nodemcu/nodemcu-firmware #1874: ESP32: need a module DHT21/22/AM2301

LDmitryL, 2017-03-22:
### Missing feature
Need the library to support reading the sensors of the type DHT21/22/AM2301
### Justification
This is the most convenient and easy way to get humidity.
And most importantly low cost!
Which is very important for home automation.
### Workarounds
API for module `dht`  from ESP8266 good

devsaurus, 2017-03-27:
I'd propose to use the RMT peripheral to implement the low-level protocol for talking to this family of sensors. An RX channel can capture data bits from the sensor in read direction and decoding to '0' and '1' is done based on the duration between the signal edges. This worked very well for the `ow` protocol.

The start impulse in TX direction can be done with simple gpio writes, roughly timed by `vTaskDelay()`.

devsaurus, 2017-03-29:
Had a look at the existing `dht` module. API seems to be ok, although I don't plan to implement `dht.readxx()` since the detection of DHT11 vs. DHT21/22 etc. appears to be unreliable in my opinion.

devsaurus, 2017-03-30:
There's a draft version available in my repo over at https://github.com/devsaurus/nodemcu-firmware/tree/dev-esp32-dht

I've tested it with an AM2302 sensor that happened to be in my stock. Checking against a DHT11 is still required before merge to upstream since this variant triggers substantially different paths in the code.

LDmitryL, 2017-04-04:
Noticed a compile error if you remove the option `UART console  auto baud detection` in `platform config` in `component config`

> CC user_main.o
> C:/msys32/home/LDL/nodemcu-firmware/components/base_nodemcu/user_main.c: In function 'app_main':
> C:/msys32/home/LDL/nodemcu-firmware/components/base_nodemcu/user_main.c:160:19: error: 'CONFIG_CONSOLE_BIT_RATE_AUTO' undeclared (first use in this function)
   cfg.auto_baud = CONFIG_CONSOLE_BIT_RATE_AUTO;
                          ^
> C:/msys32/home/LDL/nodemcu-firmware/components/base_nodemcu/user_main.c:160:19: note: each undeclared identifier is reported only once for each function it appears in
> make[1]: *** [/home/LDL/esp-idf/make/component_wrapper.mk:177: user_main.o] Error 1
> make: *** [C:/msys32/home/LDL/esp-idf/make/project.mk:386: base_nodemcu-build] Error 2

devsaurus, 2017-04-05:
Thanks for reporting @LDmitryL! I'll have a look.

devsaurus, 2017-04-05:
Running the contained functions triggers an error message in the RMT driver code. I've reported this to Espressif and suggested a fix with espressif/esp-idf#492.

Furthermore, there appear to be interoperability issues between this module and the `ow` module: Executing `dht.read*()` kills the RMT channels of the onewire protocol implementation (see espressif/esp-idf#486). This will become even worse with the `ws2812` module from #1896. I need to wait for Espressif's response until I can proceed.

@jmattsson:
What would be the best way to include the fix from `espressif/esp-ifd#486` into our code base before they ran it through their development mill? Is it acceptable to temporarily reference the `sdk/esp32-esp-idf` submodule from my PR branch? Or can I place a .patch file somewhere in our repo which is applied before compilation starts?

LDmitryL, 2017-04-06:
Yes, indeed there is a bug in the module DHT module in conjunction with the OW. Separately they work but together no.
>[0;31mE (153312) rmt: RMT[5] ERR [0m
[0;31mE (153312) rmt: status: 0x14140180 [0m
>[0;31mE (153312) rmt: rmt_set_tx_thr_intr_en(347): RMT EVT THRESH ERR [0m

the message after the call `dht.read2x(port)`. After that just reset.

devsaurus, 2017-04-06:
> Noticed a compile error if you remove the option UART console auto baud detection in platform config in component config

Fixed in `dev-esp32` branch.

devsaurus, 2017-04-14:
Module merged into `dev-esp32` branch.

zelll, 2017-04-21:
I tried to build a release version, but got
```
...
CC dht.o
D:/tmp/msys32/home/0j/nodemcu-firmware/components/modules/dht.c: In function 'ldht_read_generic':
D:/tmp/msys32/home/0j/nodemcu-firmware/components/modules/dht.c:75:29: error: 'humi' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   int humidec = (int)((humi - (int)humi) * 1000);
                             ^
D:/tmp/msys32/home/0j/nodemcu-firmware/components/modules/dht.c:74:29: error: 'temp' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   int tempdec = (int)((temp - (int)temp) * 1000);
                             ^
cc1.exe: some warnings being treated as errors
```
BTW, I did not select dht module.
```# CONFIG_LUA_MODULE_DHT is not set```

devsaurus, 2017-04-21:
I ran a build on `dev-esp32` branch just the other day without issues.
This error happened for you on this branch as well?

zelll, 2017-04-21:
I meant choosing ``` build for release``` in menuconfig, not master branch for 8266...

devsaurus, 2017-04-21:
I see.

The error was correct and I just pushed a fix for it. Strangely it's not reported for a debug build...

LDmitryL, 2017-04-24:
@devsaurus Managed to overcome the problem with RMT?

devsaurus, 2017-04-24:
Interoperability with onewire remains problematic since espressif/esp-idf#486 is still waiting for a resolution.

LDmitryL, 2017-06-02:
@devsaurus I noticed that the problem with RMT was fixed. (message `Yes, was fixed by #492, closing.` in https://github.com/espressif/esp-idf/issues/473 ). Which, in turn, referred to the problem with the ws2812 module. https://github.com/nodemcu/nodemcu-firmware/issues/1896
Does this mean that the problem is fixed and this module too?
Last update is not working with nodemcu module DHT21/22/AM2301 and OW at the same time. The system hangs without displaying any messages.

devsaurus, 2017-06-03:
No, the root cause for that is espressif/esp-idf#486.

devsaurus, 2017-07-03:
I updated to latest esp-idf which contains a fix for espressif/esp-idf#486. Waiting for a response to espressif/esp-idf#719.

LDmitryL, 2017-11-22:
@devsaurus This fix is already included?

closed 2019-01-25 by LDmitryL