nodemcu/nodemcu-firmware #1878: Problem with Cron

matthewkleinmann, 2017-03-23:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
cron.schedule("45 * * * *", function(e)
	tm = rtctime.epoch2cal(rtctime.get())
	print(string.format("*** Cron Job 45 ***  %04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
end)

printing something like *** Cron Job 45 ***  2017/03/22 21:**45**:00
### Actual behavior
Printing *** Cron Job 45 ***  2017/03/22 21:**13**:00

and if I make a cron mask for 55, that goes off at *** Cron Job 55 ***  2017/03/22 21:**23**:00
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
-- add code here

wifi.sta.eventMonReg(wifi.STA_GOTIP, function() 
 --print(wifi.sta.status())
 --print(wifi.sta.getip()) 
 
 sntp.sync("192.168.1.201",
  function(sec, usec, server, info)
    --print('sync', sec, usec, server)
	--print("IP up and Tine Synced")
	
	tm = rtctime.epoch2cal(rtctime.get())
	print(string.format("Time Sync  %04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
	
		
--------------------More stuff below here.......

  end,
  function(x,y)
   print('failed! '..x.."  "..y)
  end,
  1
 )
end)

wifi.sta.eventMonStart()

station_cfg={}
station_cfg.ssid="XXXXX"
station_cfg.pwd="XXXXX"
station_cfg.save=true
wifi.sta.config(station_cfg)

node.info()
print("Setting up Cron Jobs")

cron.schedule("0 * * * *", function(e)
	tm = rtctime.epoch2cal(rtctime.get())
	print(string.format("*** Cron Job 0 ***  %04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
end)
	
cron.schedule("15 * * * *", function(e)
	tm = rtctime.epoch2cal(rtctime.get())
	print(string.format("*** Cron Job 15 ***  %04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
end)
	
cron.schedule("30 * * * *", function(e)
	tm = rtctime.epoch2cal(rtctime.get())
	print(string.format("*** Cron Job 30 ***  %04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
end)
	
cron.schedule("45 * * * *", function(e)
	tm = rtctime.epoch2cal(rtctime.get())
	print(string.format("*** Cron Job 45 ***  %04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
end)

cron.schedule("55 * * * *", function(e)
	tm = rtctime.epoch2cal(rtctime.get())
	print(string.format("*** Cron Job 55 ***  %04d/%02d/%02d %02d:%02d:%02d", tm["year"], tm["mon"], tm["day"], tm["hour"], tm["min"], tm["sec"]))
end)

Here is some sample output:

NOTE:  I had the sntp.sync auto update switch commented out for this run.  The results are the the same with it except that it prints it's blurbs when it does time adjustments.

dofile(time.lua)  Wednesday, March 22, 2017  15:22:24

dofile("time.lua")
Setting up Cron Jobs
> Time Sync  2017/03/22 19:22:33
*** Cron Job 55 ***  2017/03/22 19:23:00
*** Cron Job 30 ***  2017/03/22 19:30:00
*** Cron Job 0 ***  2017/03/22 20:00:00
*** Cron Job 45 ***  2017/03/22 20:13:00
*** Cron Job 15 ***  2017/03/22 20:15:00
*** Cron Job 55 ***  2017/03/22 20:23:00
*** Cron Job 30 ***  2017/03/22 20:30:00
*** Cron Job 0 ***  2017/03/22 21:00:00
*** Cron Job 45 ***  2017/03/22 21:13:00
*** Cron Job 15 ***  2017/03/22 21:15:00
*** Cron Job 55 ***  2017/03/22 21:23:00
*** Cron Job 30 ***  2017/03/22 21:30:00
*** Cron Job 0 ***  2017/03/22 22:00:00
*** Cron Job 45 ***  2017/03/22 22:13:00
*** Cron Job 15 ***  2017/03/22 22:15:00
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

I believe this is from the 2.0 branch, it is from the automated build site, built with main, and the following modules: ---------- NODEMCU-MASTER-11-MODULES-2017-03-21-05-15-03-INTEGER.BIN
        modules: cron,file,gpio,net,node,pcm,rtctime,sntp,tmr,uart,wifi

### Hardware

Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.
ESP8266-01 with 8Mb or 1MB flash.  Good 3.3V power supply, generic usb/serial bridge.  Nothing at all fancy.  No I/O devices and no other hardware.  Connecting to local NTP server.

djphoenix, 2017-03-23:
Interesting. I'll take a look. Thanks for report.

matthewkleinmann, 2017-03-23:
The cron bug could very well be me doing something stupid and not a bug at
all.  Don't discount that, but I did look at it carefully, and I have been
using cron on Unix for decades.  Oddly enough I figured once I was in the
cron relm I would be on firmer ground.  Thank you for looking into this.

--Matthew

djphoenix, 2017-03-26:
There is issue in bit-shifts over 32 bits (e.g. `55-32 = 23`, `45-32 = 13`). Good point to search issue reason.

devsaurus, 2017-03-27:
Fix has been merged to `dev`.
Thanks for reporting @matthewkleinmann!

closed 2017-03-27 by devsaurus

matthewkleinmann, 2017-03-27:
Wow, that was fast!  You guys rule!  Will the fix be in the next main, and
(dumb question) how do you tell what the main you are building against is
on the automated build server?

marcelstoer, 2017-03-27:
> Will the fix be in the next main

-> https://github.com/nodemcu/nodemcu-firmware/milestone/5
-> https://github.com/nodemcu/nodemcu-firmware#releases

> how do you tell what the main you are building against is
on the automated build server?

You're building against a specific branch there. Thus, it should be fairly easy to tell by checking the branch's commit history..

PR #1884 (2017-03-27): Cron fixes
Fixes #1878.
Also fixes string-description field order as in docs.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well.
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.