nodemcu/nodemcu-firmware #1881: Document that file.mount() requires #define BUILD_FATFS

marcelstoer, 2017-03-23:
http://nodemcu.readthedocs.io/en/latest/en/modules/file/#filemount needs to tell that it requires`#define BUILD_FATFS` in `user_config.h`.

-> http://www.esp8266.com/viewtopic.php?f=23&t=14247

Also, fix the typo in this sentence:

> Besides the SPIFFS file system on internal flash, this module can also access FAT partitions on an external SD card is FatFS is enabled.

Aaaand fix the TOC at the top of the page for functions like `file.close()`.

marcelstoer, 2017-03-29:
@devsaurus your documentation states explicitly that FatFS is at R0.12a. How relevant is it to keep it up to date? Does the low version number imply anything wrt to stability or maturity? http://elm-chan.org/fsw/ff/updates.txt

devsaurus, 2017-04-19:
> How relevant is it to keep it up to date?

Most incremental fixes were related to exFat, so I check once in a while. Could be a good opportunity to update sometime soon again.

devsaurus, 2017-04-19:
PR was merged, closing.

PR #1905 (2017-04-10): File doc fix
Fixes #1881.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).

Update `file` docs with notes about differing command support for spiffs and fatfs.
Fix readthedocs' toc (hopefully).
closed 2017-04-19 by devsaurus