nodemcu/nodemcu-firmware #1882: dns problem: http.get / net.dns.resolve => dns fail

douglasjam, 2017-03-25:
### Expected behavior
I'm able to execute GET requests using `http.get` and `net.dns.resolve`

### Actual behavior
It fails since a DNS problem with code -1
```DNS fail!```
```
HTTP client: DNS failed for httpbin.org
HTTP request failed
IP: 192.168.178.48
CODE: -1
```

### Test code
```
    wifi.setmode(wifi.STATION)
    wifi.sta.config(myConfig.WIFI_SSID, myConfig.WIFI_PASSWORD)

    net.dns.setdnsserver("8.8.8.8", 0)    
    net.dns.setdnsserver("192.168.178.1", 1)

```

```
    net.dns.resolve("www.google.com", function(sk, ip)
        if (ip == nil) then
            print("DNS fail!")
         else 
            print("Resolved: " .. ip) 
        end
    end)    

    http.get("http://httpbin.org/ip", nil, function(code, data, result)
        if (code < 0) then
            print("HTTP request failed")
            print("IP: " .. myWifi.getIp())
            print("code: " .. code)
        else
            print(code, data)
        end
    end)
```

### NodeMCU version
NodeMCU-Build dev/master (2017-03-25)
modules: adc, bme280, file, gpio, http, i2c, net, node, timer, u8g, wifi

### Hardware
ESP8266 DevKit V3

douglasjam, 2017-03-26:
Via C code through arduino works well
http://shawnhymel.com/695/quick-tip-http-get-with-the-esp8266-thing/

Also tried using explicit socket but it throws error `-8`, is not something with the router, is something with lua libraries or something else =(

```
local function simulateGet()
    sk=net.createConnection(net.TCP, 0) 

    sk:on("connection", function(socket, c) 
        print("REQUESTING"
        socket:send("GET / HTTP/1.1\r\nHost: google.com\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n") 
    end) 

    sk:on("receive", function(socket, data)
        print("SUCESS!?")
    end)

    sk:on("disconnection", function(socket, errorCode)
        print("ERROR" .. errorCode)
    end)

    sk:connect(80, "google.com")
end
```

marcelstoer, 2017-03-26:
Can't reproduce, works for me (once I fixed the missing `)` in `print("REQUESTING"`). I tested both current `dev` and also the `1.5.4.1-final`.

```
REQUESTING
SUCESS!?
200	{
  "origin": "109.71.91.106"
}
Resolved: 172.217.19.164

closed 2017-03-26 by marcelstoer

douglasjam, 2017-03-26:
then I don't know what is going on, thanks for confirming, maybe a problem in the firmware of nodemcu-build.com

marcelstoer, 2017-03-26:
Nope, I eat my own dog food and use binaries from there.

karrots, 2017-03-26:
Couple of items.

1. Your if statements don't need ()
2. Are you calling the HTTP and DNS functions from the GOTIP wifi event (or at least after)? 

You may be attempting a network operation before the ESP is ready for such an operation. Try this code and see if it works better.

[https://gist.github.com/karrots/b5dc4151164945f4dc36f217ec904732](https://gist.github.com/karrots/b5dc4151164945f4dc36f217ec904732)

marcelstoer, 2017-03-26:
Thanks Jonathan, I silently assumed the OP had the basics covered. @douglasjam study the Gist or/and or have a look at http://nodemcu.readthedocs.io/en/latest/en/upload/#initlua.

ricardoas30, 2018-10-22:
Hello, I would like to know if you could help me. I am having a request problem on a post call. the error is as follows: HTTP request failed
follows the requisition:

```
http.post('https://api.pushbullet.com/v2/pushes',
  'Content-Type: application/json\r\n\Access-Token: TOKEN_DO_PUSHBULLET\r\n',
  '{"body":"my body","title":"my title","type":"note"}',
  function(code, data)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, data)
    end
  end)
```

> NodeMCU custom build by frightanic.com
> 	branch: master
> 	commit: XXXXXX
> 	SSL: false
> 	modules: file,gpio,http,mdns,mqtt,net,node,pwm,tmr,u8g2,uart,wifi,wifi_monitor
>  build created on 2018-10-21 18:19
>  powered by Lua 5.1.4 on SDK 2.2.1(xxx)

Thank you

nwf, 2018-10-22:
@ricardoas30  Thread necromancy of completely unrelated issues is deeply unappreciated.  For support options, see https://nodemcu.readthedocs.io/en/latest/en/support/ .