nodemcu/nodemcu-firmware #1883: Rewrite of the Lua DS8B20 module

TerryE, 2017-03-26:
I am not sure at this stage whether this is a bugrep or a missing feature.  I am using DS18B20s in anger and decided that I'd rewrite the ds18b20 lua module.  My reasons were as follows:

-  I wanted to make the module tmr compatible, and to this end I've split the read function into a `convert_T()` call which issues the convert command and a separate `read()` which collects the response from a single thermometer which and be collected on a ALARM_SEMI callback.
-  The module now supports multiple DS18B20s correctly in that a `convert_(true)` does an `ow.skip()` instead of an `ow.select()` and starts all DS18B20s converting in parallel.
-  The module saves the search output to `ds18b20_save.lc` and reloads this subsequently on first use so that you don't need to do the search each waking after deep sleep.
-  I've also used Lua best practice for locals and upvalues to minimise size / runtime.

Here is my test harness which shows the difference at an application level:
```Lua
local ds = require("ds18b20")
local to_string, print, gpio2 = ds.to_string, print, 4
local t1, addrs

ds.init(gpio2)
addrs = ds.get_addrs()

if addrs and #addrs > 0 then
  print("Total DS18B20 sensors: ")
  for k,v in ipairs(addrs) do print (' ',to_string(v)) end
  local tmr1,tmr2 = tmr.create(), tmr.create()
  tmr1:register(1000, tmr.ALARM_SEMI, function()
    local t = {}
    for i = 1, #addrs do t[i] =  ("%8.3f"):format(ds.read(addrs[i])) end
    print(unpack(t))
  end)
  tmr2:alarm(30000, tmr.ALARM_AUTO, function()
    ds.convert_T(true)
    tmr1:start()
  end)
else
  print("No OW thermometers found")
end
```
You can happily crank this up to collect half a dozen probe temperatures each second.

TerryE, 2017-03-31:
@devsaurus @pjsg @jmattsson I've added a gist copy [here](/TerryE/5ea4f51492468a46b095279b0fff6935).  I'd be interested in your views.  I am rewriting a few of these for my own use anyway, and I'd like your views on whether it is worth replacing the existing libraries as we do so.

What I've done here is to make the code a lot more runtime efficient as well as address a few functional holes.

karrots, 2017-03-31:
Gist link needs to be fixed. Github is improperly prepending `github.com` instead of `gist.github.com`

TerryE, 2017-03-31:
Sorry Jonathan.  Done :smiley:

jmattsson, 2017-04-03:
Haven't used any DS18B20s with NodeMCU, so have nothing to compare against, but this seems good?

TerryE, 2017-04-03:
Why the ? :smiley:

jmattsson, 2017-04-03:
Because I don't feel like I'm an authoritative source on such matters? :)

johndoe71rus, 2017-04-22:
try test your code. But get 

> dofile("ds18b20_terry.lua")
No OW thermometers found
> 

I use nodencu pin3(GPIO0)
old example, is work for me.

vsky279, 2017-04-22:
@TerryE I have upgraded my module using your suggestions. It should fix majority of flaws discussed in https://github.com/nodemcu/nodemcu-firmware/pull/1820.

Please have a look here: https://github.com/vsky279/nodemcu-firmware/tree/DS18B20/lua_modules/ds18b20

Enhancements:
+ primarily targetted to float rather than integer version of Lua
+ upvalues are used much more than in previous version - it should lead to lower memory usage and more efficient code
+ `ow.skip()` now supported - when only non-parasit powered sensors are present on the bus the conversion is started for all sensors at once
+ bus search performed in tasks - no long loop - no need to use `tmr.wdclr()` (though I'm not sure it was even needed - this hase been taken it from the previous version of the module)
+ no more use of `encoder` module
+ `enable_debug` method implemented for debugging purposes
+ no need to perform search before each readout (can be invoked optionally by `search` parameter). Sensor addresses are not stored on the flash - I find it more easier to perform search rather than writing to flash (though your method is very elegant). Storing `ds18b20.sens` should be done by the application if deep sleep friendliness is requested.

Possible enhancements:
+ still using the 750 ms delay for conversion - I have no issues with this in my usecase (even mixing parasitic and powered sensors)
+ only centigrade implemented for the time being
+ no family search supported

If there is no opposition I would submit a PR as this implementation is better than the existing module. I still think the module should support both parasit and non-parasit powered sensors (which is not supported by your module).

TerryE, 2017-04-25:
@vsky279 Lukáš, I thimk that it would be a good thing to converge on a single implementation that we both support.  If we can do that, then it will be a good solution.  Things that I would prefer to add:
-  Search
-  Conversions other than C.  Our US colleagues seem wedded to °F and inches, etc.
-  _Optional_ storing and retrieval of address to flash
- I don't like commented-out code (the integer stuff).  I need to suggest a better solution.

Give me a few days.  I am up to my neck in my core ro TSring stuff. :)

vsky279, 2017-04-26:
@TerryE I've started to implement your suggestion (except the integer stuff :-) ) so don't spend your time on it. I will come with a proposal that you can review afterwards.

vsky279, 2017-05-22:
@TerryE I need to restart. What I miss is the optional storing and retrieval of address to flash.
As for the integer stuff I let you suggest better solution when I'm done.

vsky279, 2017-05-26:
@TerryE Please have a look at the latest version of updated module: https://github.com/vsky279/nodemcu-firmware/blob/DS18B20/lua_modules/ds18b20/ds18b20.lua

It has all the functionalities discussed. The documentation is not yet updated. Please see the example in the same directory.
But I think it could be more efficient. It should avoid probably addressing the `sens` table by indexes `addr`, `parasite` and `status` and use a bit different approach.

Please let me know your thoughts about it.

vsky279, 2017-05-30:
Done. I need to update documentation and will submit a PR.

closed 2018-02-11 by TerryE